paparazzi/paparazzi #1064: Google DCMA takedown request

eric-wieser, 2015-01-10:
FYI [this DCMA takedown request](https://www.chillingeffects.org/notices/10281020) lists this paparazzi  github repository [as one of the infringing urls](https://www.chillingeffects.org/notices/10281020#infringing_url_601432202).

Just in case you stop being indexed by google in future, and have no idea what's happening.

flixr, 2015-01-10:
Thanks for the info!
Obviously this is totally bogus as there are no movies in this repository...
Any idea on how to make sure this does not happen?

eric-wieser, 2015-01-10:
Fix DCMA laws?

Probably because some company made a film with the word paparazzi in the title. I'm afraid you're going to have to stop using real words...

eric-wieser, 2015-01-10:
Amusingly they also issued a takedown request for an issue - antirez/redis#1651

benlaurie, 2015-01-11:
On 10 January 2015 at 14:31, Felix Ruess <notifications@github.com> wrote:

> Thanks for the info!
> Obviously this is totally bogus as there are no movies in this
> repository...
> Any idea on how to make sure this does not happen?
>

As Eric says, you need to fix the DMCA laws. :-)

More helpfully: if there is any negative impact, I work for Google - let me
know and I'll try to get it fixed.

OpenUAS, 2015-01-14:
Read some more here
https://gist.github.com/Psycojoker/69c00d0d11f26a46ac93

OpenUAS, 2015-01-14:
@benlaurie , yes any help to let this madness stop welcomed.

closed 2015-03-02 by flixr