paparazzi/paparazzi #1075: It is not possible to redefine the HOME block

alonsoac, 2015-01-27:
If I try to make my own HOME block I get  Error in flight plan: Block 'HOME' defined twice

The following patch makes a change so that the default HOME block is only added if the user did not include his own.

--- <html>gen_flight_plan.ml (<b>Yesterday 5:47:08 PM</b>)</html>
+++ <html><b>Current File</b></html>
@@ -742,6 +742,13 @@
   in
   (sector_name, List.map p2D_of (Xml.children xml))
 
+let add_home_block = fun blocks ->
+  let found = ref false in
+  List.iter (fun b ->  if (ExtXml.attrib b "name" = "HOME") then found := true ) blocks;
+  if (!found = false) then
+    blocks @ [home_block] 
+  else
+    blocks
 
 (************************** MAIN ******************************************)
 let () =
@@ -765,7 +772,7 @@
     let xml = Fp_proc.process_relative_waypoints xml in
 
     (* Add a safety last HOME block *)
-    let blocks = Xml.children (ExtXml.child xml "blocks") @ [home_block] in
+    let blocks = add_home_block (Xml.children (ExtXml.child xml "blocks")) in
 
     let xml = ExtXml.subst_child "blocks" (index_blocks (element "blocks" [] blocks)) xml in
     let waypoints = Xml.children (ExtXml.child xml "waypoints")

gautierhattenberger, 2015-01-28:
The purpose of HOME block at the end is to be sure that you have something to do if you reach the last block (and last stage) without protection (endless stage, deroute,...).
So removing it is not an option.
We could change the name, but I don't see the real benefit. Currently HOME waypoint is mandatory and HOME block is reserved...

alonsoac, 2015-01-29:
The code is not meant to remove the HOME block, only to avoid duplication when it already exists in the user's flight plan. The code could be changed to make sure that it is always at the end to give that protection.

What I really wanted to do is customize the HOME block so that when entering this mode the a/c keeps the current altitude and only when it is close to HOME it descends to the defined HOME altitude.  The a/c might be flying higher because of obstacles between its current position and the HOME position, so if it descends immediately it could crash.

Other users might have their own wish with respect to the HOME mode procedure so it is useful to be able to customize it.

OpenUAS, 2015-01-31:
alonsoac, IMHO it is better to your your own block that act the way you want it an not alter default HOME, an exeption in your flightplan could do that. It is reservers as @gautierhattenberger already remarked this has a deeper reason for UA validation of default behaviour.

alonsoac, 2015-02-01:
I am happy with being able to make my own HOME block. If in the future someone else wants to do it hopefully they will find the code above useful. For the moment lets close the issue.

closed 2015-02-01 by alonsoac