paparazzi/paparazzi #1077: less important elements are shown at the top while blocks are at bottom in flight plan xml tree view

alonsoac, 2015-01-27:
It looks much better and is easier to use if we add the elements in reverse with blocks at top

<PRE>
--- <html>xmlEdit.ml (<b>Yesterday 5:47:08 PM</b>)</html>
+++ <html><b>Current File</b></html>
@@ -167,7 +167,7 @@
   let store = GTree.tree_store cols in
   let row = store#append () in
   set_xml store row xml;
-  List.iter (fun x -> insert_xml store row x) (Xml.children xml);
+  List.iter (fun x -> insert_xml store row x) (List.rev (Xml.children xml));  (* reverse so blocks show on top *)
   store;;
</PRE>

flixr, 2015-01-27:
@alonsoac if you use the pull request functionality it makes things much easier...

alonsoac, 2015-01-28:
unfortunately the fork that I have in github is not linked to the original paparazzi repository so I haven't figured out how to send pull requests from that one.

gautierhattenberger, 2015-01-28:
I think the order of the block in the flight plan are important (relative to the DTD definition). It also makes sense to define waypoints or global exceptions before the block instructions.
We can probably reverse only the display, but I'm afraid it will be more disturbing to have the xml file and editor display reversed. Also, you can collapse the first elements if not needed.
So, not really in favor of changing this.

flixr, 2015-01-29:
Also not in favour of changing it, header and waypoints are colllapsed by default anyway...

alonsoac, 2015-01-29:
The intention here is to make the GCS easier to use and part of that is removing or moving to less prominent positions the elements that are not really useful during the flight operation.

Specifically with the flight plan tree:
- the header element doesn't seem to be useful at all. All the header is compressed in one line and just the beggining is seen as there isn't even a way to scroll to see the entire line. I do not see why the operator would be interested in this information during flight.  This could be hidden with no impact for the flight operator.
- the sectors element is also not usefull since sectors can be seen on the map. This could be hidden with no impact for the flight operator.
- the waypoints can also be seen in the map and since the positions are not updated in the tree when the waypoints are moved the information does not even reflect the correct positions. This could be hidden with no impact for the flight operator.
- exceptions is the only one that I think might be useful sometimes as the flight operator might need to check them. This is less used and could be located at the bottom of the tree.
- blocks are the most used element and should be given the most prominent position

flixr, 2015-01-29:
Hiding some in the GCS tree view would be fine with me (if that is easily possible), but changing the order doesn't really make much sense to me, as it would also be written back to the xml in that order...

gautierhattenberger, 2015-01-29:
And it should not go into xmlEdit. There is already a pretty printer somewhere, it may be improve to remove some xml node, not only arrange attributes.

OpenUAS, 2015-04-29:
FYOI: The sectors element is useful, for some(me) if we fly without a map from a small screen list based GUI only.Waypoint get highlighet so you know when reached... . etc. etc. 

But my different use is of no benefit to all ;). So thinking of it... an idea is to add a flag with visible or not visible for a block, (or even waypoint?). Other ideas if considered add it to this issue.

agressiva, 2015-04-29:
I like too much the sectors. I even use it at the field.
i think a good idea is to use KML as sector.
Everytime i go to the field i need convert the KML to sector.

podhrmic, 2017-08-26:
@flixr @gautierhattenberger should we close this as "not an issue" or for feedback timeout?

gautierhattenberger, 2018-06-06:
closing as won't be fixed in current GCS
see https://github.com/paparazzi/paparazzi/projects/2

closed 2018-06-06 by gautierhattenberger