paparazzi/paparazzi #1079: cleanup UDP interface/implementation

flixr, 2015-01-29:
We have:
- arch/linux/mcu_periph/udp_arch.[ch]
- fms/fms_network.[ch]
- modules/computer_vision/lib/udp/socket.[ch]

fvantienen, 2015-02-03:
From the top of my head only "arch/linux/mcu_periph/udp_arch" is used now in the airborne code. Except for some computer vision modules. Fms can be removed totally if I'm correct.
I think the "arch/linux/mcu_periph/udp_arch" is the way to go, because it has the same interface as in UART etc. So you can switch between those 2 easily.

Next to this we should also check own socket implementations, which don't use any of these and replace those to make it nicer.

flixr, 2015-04-14:
@fvantienen fms_network is still used in natnet2ivy...
I tried to convert to use udp_socket instead with https://github.com/flixr/paparazzi/commit/638fc0f2a7fcef3cd23c3df6780497f90346c963, would be nice if you could confirm that natnet2ivy still works...

OpenUAS, 2015-04-29:
Needs testing in Cyberzoo. Will re-assign or report if not working

flixr, 2015-06-09:
should be finally closed with merge of #1162 which removed the fms stuff

closed 2015-06-09 by flixr