paparazzi/paparazzi #1084: Datalink information in GCS bug when restarting server

fvantienen, 2015-02-03:
Sometimes when you restart the server(or datalink), instead of the timeout I get a 2/2 in the datalink block of the GCS. This means that it has connection with 2 datalinks, which I never had. So the GCS shows wrong information.
When restarting GCS this "2/2" is removed and everything works fine again.

gautierhattenberger, 2015-03-03:
I can't reproduce the bug (using an ardrone2). Do you have a reproducible sequence to make this fail ?

flixr, 2015-06-18:
@fvantienen I was also unable to reproduce this and have never had this happen so far.
So unless you have some more information on this topic or steps to reproduce it, I suggest we close this issue.

fvantienen, 2015-06-24:
I haven't seen this anymore.. not sure why it happened.
Will close it for now, and if it happens again will reopen ;)

closed 2015-06-24 by fvantienen