paparazzi/paparazzi #1085: Paparazzi center generated double messages

OpenUAS, 2015-02-03:
If one just want to test compile/build half way creating a new airframe file just to quicly validate, one could get an message in the Paparazzi centertext area like:

  File not found: /home/theuser/paparazzi/conf/modules/ins_chimu_spi.xml

So far all dandy... since we made a mistake, that is not what this issue is about, it is about:

One also gets the **same message** output on the commandline the first time one runs clean/build. If one then cleans and build again, the message is not there anymore.

If you close paparazzi center, restart vai ./paparazzi in your terminal , do a clean uild , again once the message on the terminal line also

Expected is not to have any text added behind ./paparazzi in the terminal, since this can cause unwanted behaviour.

flixr, 2015-05-04:
I cannot reproduce this, never get an error on clean...

flixr, 2015-05-04:
Ah, misunderstood you... it seems that if the aircraft config is reparsed (e.g. upon selecting the aircraft at startup or after selecting another one) it will print the error to the console where paparazzi_center has been started (pc_aircraft.ml line 187 and 300).
We could simply remove printing these errors, but I'm not quite sure if we really want to not inform the user about this... @gautierhattenberger what do you think?

gautierhattenberger, 2015-05-04:
It is probably not nice to hide errors. If we want to make it better, the complete workflow of build and message has to be checked and updated. I don't really have the time to do it myself right now.

flixr, 2015-05-04:
So we close this as WONTFIX?

OpenUAS, 2015-11-20:
OK, it still in issue history after all.

closed 2015-11-20 by OpenUAS