paparazzi/paparazzi #1093: Problem with waypoint altitude

agressiva, 2015-02-10:
I am testing the autoland using MentorEnergy AC.

the AF has alt=30 but when geo_init run it set to -155m when the correct is ground altitude + 30 (215m)
The same occur with TD. After geoinit it get -185 when the correct is ground altitude + 0 (185m)

 i think someone changed a signal in the code becahuse some time ago it worked fine

flixr, 2015-02-10:
Are you sure you specified the waypoint correctly? You probably want to use height instead of alt...

Otherwise, can you please provide the exact paparazzi_version and the flight plan you are using?

agressiva, 2015-02-10:
Felix, i am using v5.5_devel-222-ge3edd57-dirty   with default mentorenergi AC and versatile_airspeed flight plan.

I changed the "alt" by "height" and it worked fine as you tell.

some flightplane have the altitude set to 215 and 185 but i think a good idea use heigh on all flight plan.

thanks for help.

Date: Tue, 10 Feb 2015 10:21:24 -0800

flixr, 2015-02-10:
Altitude is "absolute" so over sea level. If you wp alt is at e.g. 230m and flight plan alt at 200m, the waypoint will still be 30m over ground (difference between the two) even if you relocate the whole flight plan to another location altitude via the GeoInit block...
height is just a "shortcut" for alt=ground_alt+height

Closing this then, as it is not a bug...

closed 2015-02-10 by flixr