paparazzi/paparazzi #1098: Datalink ID start value

OpenUAS, 2015-02-11:
The current message of a new link detected starts with index 0. From programmers perspective quite OK. Various ground station operators however where somewhat confused, to put it mildly.

1) Rather start at 1. aka the first link.... 

2) And adding in ID in front of the value makes it much clearer specifically when speech is enabled.
and the "new link detected" could do without the New, specifically when speech is enabled it should be as short.

The preference would be:

Datalink detected with ID X

or maybe even better:

Datalink X connected

With multiple links this still sounds OK like

Datalink 1 connected, Datalink 4 connected Datalink 3 connected

Although ith multiple links one could ague they are only  detected and not used as backup link thus not "Connected", but that discussion can land up in another issue.

flixr, 2015-03-09:
@OpenUAS if you are happy with the solution of @gautierhattenberger (added to master with 4e378b3), we can backport it to v5.4 and then close this issue...

flixr, 2015-03-16:
Feedback timeout...
Backported to v5.4 with 80cc06b76c2fa33099b32db2a1c7479a0a4a497b

closed 2015-03-16 by flixr