paparazzi/paparazzi #1100: takeoff/speed detected with a/c on ground and thottle killed

alonsoac, 2015-02-12:
There are some problems with takeoff detection in fixedwing/main_ap.c:665
- it is not clear what purpose this serves
- it is not possible to deactivate this detection or redefine the speed at which takeoff is detected
- takeoff is detected even if throttle is at 0%
- incorrect takeoff detection can cause other issues for conditions that take place when autopilot_flight_time>0. For example, going into HOME mode when RC is lost.

It is possible for stateGetHorizontalSpeedNorm_f to return a value > 5 when a/c is indoors with poor GPS reception because of large sudden changes in reported position. This causes the incorrect takeoff detection.

alonsoac, 2015-02-12:
There is a very dangerous situation when:
-a/c is on the ground or bench
-flight plan is on a block that keeps the throttle at 0% but does not set kill_throttle to 1. This would appear to be safe.
```
<block name="StandStill">
  <attitude roll="0" throttle="0" vmode="throttle"/>
</block>
```
-a/c is too far from home (really or because of bad GPS position)
-bad GPS position is detected as speed and takeoff is detected, launch is to to TRUE
-in fixedwing/main_ap.c:388 mode is changed to HOME. 
-this causes the motor to start unexpectedly

OpenUAS, 2015-03-03:
@alonsoac love to help out with the issue but fully understanding it was not really easy...

Things like "takeoff is detected with 0% throttel" Why would this not be valid, I launch motorized gliders often on a hillside with no throttle. only below a certain airspeed or height the motor kicks in and the flightplan makes that it goes up +100m. It then glides an repeates, until battery is below a level then it autolands.

Why I appricaiate you concerns, all issues you mentioned, can be solved with a good (custom) flightplan IMHO

alonsoac, 2015-03-09:
I think takeoff detection should be activated when needed by people who know what they are doing and the risks involved with automatic detection.
I don't have a problem with this because I know what I am doing but I think this might be an issue for new users.

saaib, 2015-03-09:
I can see this feature disabled by default and provide an airframe 
property to enable it, so users enabling it are expected to know what to 
expect.

gautierhattenberger, 2015-03-10:
I agree the takeoff detection could be made more robust (especially to reject false detection), but I don't think it should be disabled.
Several control or safety systems have different behaviors when on ground and in flight (trigger HOME, GPS_LOST, reset control loops,...) and if the `launch` variable or any `in_flight` detection is not automatic, people will forget it someday (whatever their experience is), causing even more troubles.
`launch` and `in_flight` can probably be two separated variables, but at least one should be automatic.
Any idea for a set of robust conditions to detect that a plane is flying are welcome.

gautierhattenberger, 2021-12-13:
timeout

closed 2021-12-13 by gautierhattenberger