paparazzi/paparazzi #1101: RC_LOST_MODE should not be set if kill_throttle==1

alonsoac, 2015-02-12:
in fixedwing/main_ap.c:392

if throttle is killed there is no benefit to setting the RC_LOST_MODE. It would be better to simply go to AUTO2 ?

flixr, 2015-02-12:
> if throttle is killed there is no benefit to setting the RC_LOST_MODE

Why not? By default RC_LOST_MODE is HOME mode

> It would be better to simply go to AUTO2 ?

Don't think so, you will have no idea what state and block of the flightplan you are currently in or what the next waypoint might happen to be...

alonsoac, 2015-02-13:
Not sure if I am missing somthing byt if the throttle is killed it doesn't matter what mode it is in, it will just fall to the ground?

The way I use paparazzi I first set a block of the flight plan which I am confortable with while the a/c is controlled in MANUAL or AUTO1. For example the StandBy block.  This way I know that if for any reason the pilot gets in trouble in manual he can just switch to AUTO2.
But again if the throttle is killed it will still fall no matter what block it is in right?.

flixr, 2015-02-13:
Sorry, I don't understand what you are getting at...
If throttle is killed, it will - well - kill the throttle. If you have an airplane and get into RC_LOST_MODE (which defaults to HOME), it will still try to fly towards the home waypoint, just without throttle...

flixr, 2015-02-20:
@alonsoac could you please clarify?

alonsoac, 2015-02-25:
I was actually thinking about an a/c which is on the ground with throttle killed and RC is lost.  But yes it could also happen in the air and then it is useful to go to home mode.

closed 2015-02-25 by alonsoac