paparazzi/paparazzi #1105: RT Paparazzi fully integrated, offering new possibilities with failsafe tasks, data logging and autopilot monitoring

podhrmic, 2015-02-13:
Splitted items from wishlist #620 (as recommended by @OpenUAS)
(For Paparazzi 6.0)

flixr, 2015-02-13:
Are more specific things that you can list here?
Personally I don't see the advantage of having this issue as long as we still have #560 open....

podhrmic, 2015-02-13:
Again, my bad - haven't noticed the original issue. Will work with #560 then.

flixr, 2015-02-13:
Ah, sorry... that is the old pull request.
We can use this issue to track progress of a proper chibios integration.
Please edit the detailed issue description and add a list with subtasks that we can then tick off...

gautierhattenberger, 2015-03-12:
The current chibios arch layer seems to only use synchronous and blocking calls on mcu_periph (uart, i2c, spi), which will only work if the autopilot is correctly separated into tasks. This will probably not be the case before a long time and it makes this branch not really usable at the moment for a complete autopilot.
Is their any objection to create one task per mcu periph (uart1, uart2, i2c1, i2c2, spi1, ...) to handle its activity, and one task for the rest of the autopilot (i.e. the arch independent part) ?

flixr, 2015-03-12:
As long as you can safely "write/read" data from the peripherals from other threads (i.e. the main thread running all the autopilot code for a start).

podhrmic, 2015-03-12:
We already have a code with separated autopilot tasks and we are using it for autonomous flights. In this branch I would like to merge that code into newest master. Right now I am polishing the arch drivers, which takes some time. Imho creating threads for mcu periphs and one for rest of the autopilot negates any benefits RT-OS has.

The case when more than one task has to access the same peripheral is already handled in the arch driver (mutex blocks in case of i2c or spi). For serial ports this shouldn't be an issue, since more threads can write into a port, but just one thread can read.

gautierhattenberger, 2015-03-12:
This is my point, it only works if the rest of the autopilot is using threads, which is not the case yet. So not really usable at the moment. The overhead of having one thread per periph is probably not that big, and it allows you to use the autopilot in it's current state but also with an RTOS. We will just wait for another thread to finish instead of waiting for the return function of a submit for instance.

podhrmic, 2015-03-12:
true, but I dont see any benefit of using RTOS like that. Or why would you use it in such way?

podhrmic, 2015-03-12:
@gautierhattenberger just noticed your message in gitter. The test work with "regular" ChibiOS arch, I never tried Apogee board.

flixr, 2015-03-12:
@podhrmic because then we could start putting things into separate task step by step...
Start with one that runs it all, then put some failsafe checks in a separate one, then sensor reading, etc...

gautierhattenberger, 2015-03-12:
I want to get rid of the chibios-libopencm3 integration and use a plain chibios integration because it makes things easier.
The purpose of chibios-libopencm3 is to have logging on SD card on apogee. So we can help but it as to work just like what we add before at a first step, which is one thread for logging, one thread for autopilot.
It will take a long time before we can do that with the current implementation. And it can be change later when the complete RTOS thing is working.

podhrmic, 2015-03-12:
OK, got it. 
SPI is already thread safe, see: https://github.com/paparazzi/paparazzi/blob/rt_chibios/sw/airborne/arch/chibios/mcu_periph/spi_arch.c#L249

I2C is also already thread safe, see: https://github.com/paparazzi/paparazzi/blob/rt_chibios/sw/airborne/arch/chibios/mcu_periph/i2c_arch.c#L119

Serial port: 
writing is thread safe, since it just puts a byte in the queue: https://github.com/paparazzi/paparazzi/blob/rt_chibios/sw/airborne/arch/chibios/mcu_periph/uart_arch.c#L149

Note you can use "transmit buffer" function: https://github.com/paparazzi/paparazzi/blob/rt_chibios/sw/airborne/arch/chibios/mcu_periph/uart_arch.c#L163
But performance-wise it doesn't make much difference unless you're sending *a lot* of data (and then you should DMA buffered driver anyway)

Reading has to be in a separate thread, since there is a wait for the received data, see: https://github.com/paparazzi/paparazzi/blob/rt_chibios/sw/airborne/arch/chibios/mcu_periph/uart_arch.c#L173

Radio input needs its own thread. The rest should work in one giant thread, although timing might not be great.

flixr, 2015-03-12:
> Serial port:
writing is thread safe, since it just puts a byte in the queue: https://github.com/paparazzi/paparazzi/blob/rt_chibios/sw/airborne/arch/chibios/mcu_periph/uart_arch.c#L149

This is exactly where I mean we have to be careful... while the single function call might be thread safe, we have to keep in mind that when we have use functions that use this internally (e.g. pprz_send_msg_X) it is not "thread safe" any more... as multiple simultaneous calls will end up interleaving the bytes of two messages instead of adding each of them "en-block".

flixr, 2015-06-09:
@gautierhattenberger what is up with the rt_chibios branch? It seems that AHB_CLK is missing for lisa_mx_rt board and there are some other compile errors for apogee...
Would be great if we could get those fixed asap so that we can merge the rt_chibios branch...

flixr, 2015-06-09:
Hopefully correctly set AHB_CLK for lisa_mx_rt, sw/airborne/firmwares/fixedwing/main_chibios.c seems to be missing for apogee_chibios

gautierhattenberger, 2015-06-09:
True, I forgot this file.
It seems to start working, what is really missing is the level of "configurability" that we currently have with libopencm3. And also finishing the SD card modification.
So I'm not sure we can merge this for the next release (assuming it will be before the summer...).

podhrmic, 2015-06-09:
I don't think it should be merged as it is. I will have time to work on HAL integration at the end of June. It would make sense to merge once the HAL is included, so people can play with Chibios, but indeed that depends on the timeline for v 5.6 - when do you want to release it?

gautierhattenberger, 2015-06-09:
it was planned for July 6th...

podhrmic, 2015-06-09:
Hmm, a tight schedule but doable...

gautierhattenberger, 2015-06-09:
sorry, I meant June 6th. But even July will be difficult!

flixr, 2017-03-04:
Closing this, since we added chibios support with #1699 and follow ups...

closed 2017-03-04 by flixr