paparazzi/paparazzi #1113: update geomag model to latest WMM2015

flixr, 2015-02-18:
Update the World Magnetic Model from 2010 to 2015 which is now available: http://www.ngdc.noaa.gov/geomag/WMM/

OpenUAS, 2015-03-03:
I did not dare to ask ;) but yes, please very welcomed!

flixr, 2015-04-29:
Major? Really??

OpenUAS, 2015-04-29:
Issue assigned to self...And no good autoflying without good data and same fixed firmware (need location set)

OpenUAS, 2015-04-29:
Having the 2015 WMM.COF file busy converting to 2015 discover in 2010 current code defines...
At first glance the first 0.0 looks wrong(Swapped).. ongoing.

-----------
...
const double gh1[MAXCOEFF] = {
  0.0, -29496.6, -1586.3, 4944.4,
  -2396.6, 3026.1, -2707.7, 1668.6, -576.1,
...
const double gh2[MAXCOEFF] =  {
  0.0, 11.6, 16.5, -25.9,
  -12.1, -4.4, -22.5, 1.9, -11.8,

The table:
   2010.0            WMM-2010        11/20/2009
  1  0  -29496.6       0.0       11.6        0.0
  1  1   -1586.3    4944.4       16.5      -25.9
  2  0   -2396.6       0.0      -12.1        0.0
  2  1    3026.1   -2707.7       -4.4      -22.5
  2  2    1668.6    -576.1        1.9      -11.8
...

OpenUAS, 2015-07-22:
Looking at NASA table comparing 2010 data, could it be the first index values where swapped all along ?

OpenUAS, 2015-07-29:
https://github.com/paparazzi/paparazzi/pull/1270 If pull not honored, will reopen issue

closed 2015-07-29 by OpenUAS