paparazzi/paparazzi #1163: Camera hardware support for Parrot Bebop

OpenUAS, 2015-04-29:
This meta issue is to track progress of the Parrot Bebop support for the build in camera hardware
 
- [x] bottom camera
- [ ] front camera
- [ ] H264 video encoder
- [ ] GPU

OpenUAS, 2015-11-20:
Any new insights @fvantienen , if so plz assign it to milestone 5.8 if it is feasable else v6.0, but that would be less fun ;)

dewagter, 2015-11-22:
- GPU is supported, but what do you want to run on the GPU?
 - ISP and H264 require quite some library files.

flixr, 2016-07-05:
So ISP is implemented with #1750 thanks to @fvantienen

OpenUAS, 2017-02-02:
Big fat **THANKS** to @fvantienen for adding all the magic by sleepless nights spells. Superb! The next step, separation and refactoring is already ongoing for an even cleaner solution

closed 2017-02-02 by OpenUAS