paparazzi/paparazzi #1167: Paparazzi center minimum size too restrictive

flixr, 2015-05-02:
On small screens the minimum size (width) of the paparazzi center is way to restrictive (too wide).

It should be possible to resize it to a smaller width, but somehow this limited by the default width of the combo boxes and buttons...

OpenUAS, 2015-11-20:
Lets say 800x600 is our minimum?(or 640x480 then we can go from there to define what "fixed means". Ye in the past I'v tested it on a RPi with touch 320x240 indeed problematic ;)

flixr, 2015-11-23:
The problem is not choosing a min size... some columns can be made infinitely small. But the right column with target/flashmode/settings drop downs and buttons can't be made small enough.
Not sure how to fix that...

hooperfly, 2015-11-24:
We could emulate, to some degree, how modern website UI degrades on various device(i.e. a full screen browser on a computer/laptop down to smartphones and all aspect ratios and screen sizes in between.) Usually they degrade into vertically stacked containers. We could also leverage the `layout` technique we use with the GCS to specify a layout model, so to speak for the Paparazzi Center. Currently the Paparazzi Center looks like the following:
[image]

A first order degradation model might be to show the menu bar, followed by the config panel area, followed by the build/execute panel, followed by the status bar.

gautierhattenberger, 2018-04-12:
the python version can be used and should be smaller

closed 2018-04-12 by gautierhattenberger