paparazzi/paparazzi #1177: GMaps fail to load in simulator/Need location of code that loads map

jroboticsgroup, 2015-05-10:
I am relatively new to paparazzi coming from APM. I have an issue with the GCS and the loading of google map tiles which dont load at all, all other map types load apart from it.

Another issue I noticed in the simulator is that after an aircraft has landed trying to go airborne immediately(clicking "launch") will send the command to the system but the map will remain static and unchanging(you can see the command in the flight console), I am unsure why this is happening.

I am looking to do some modifications to the GCS including the addition of a terrain map feature which shows height of terrain relative to the UAV as well as the integration of other maps such as wikimapia, plus some major GUI restructuring. 

Would love to get some assistance in this arena.

Regards

pesimov, 2015-05-10:
Hi, 

The Google map tiles issues has occurred to me as well. It happened when I switched offices. I assumed that people had downloaded too many maps. 

Are you running the simulator for the first time? I found out that I had to wait for the GPS-wait block to be finished first. This occurs when the altitude does not change and matches initial position. 
Then, I click on the 'landing strip' button first, and then click on the 'Launch' button. The 'landing strip' is on the very left corner and the 'launch' button is the green airplane flying. 

I do not have enough experience with paparazzi so far, so it might be better for someone else to comment on GUI changes. 

:)

Best.

jroboticsgroup, 2015-05-10:
I ran it several times. I will try your suggestion.

The GCS has a lot of features but it is very clunky, I have already started changing the icons but I want to make much greater UI mods to get it in line with some of the other GCS out there including mission planner and QGroundstation. 

I think the terrain feature would be excellent as it will allow the GCS to alert persons of hills or other highlands that poses a threat to the UAV at specific flight altitudes.

gautierhattenberger, 2015-05-10:
Hi,

The basic simulator ("sim" target for fixed-wing) will end when your 
plane goes below the ground level. Then you just need to restart the 
simulator agent in the paparazzi center. Historical reason is that it is 
not very realistic takeoff again after a landing without reseting the 
autopilot.

Concerning the height over ground, it is already in the GCS, you just 
need the proper SRTM data. If you enable it in the "Nav" menu, it will 
try to download it if not available. Then the ground alt under the mouse 
will be displayed in the top right corner, and the plane AGL in the 
strip. See http://wiki.paparazziuav.org/wiki/GCS for available GCS features.

For the GMaps loading not working, we would need more info to help you. 
What is your paparazzi version (run ./paparazzi_version in a terminal) ? 
What is the content of the file conf/maps.xml ?

What kind of new features would you like to see in the GCS ?

Gautier

jroboticsgroup, 2015-05-10:
Hi Gautier, thanks for getting back to me.

Thanks for clearing that up with the simulator. 

In regards to the height indicator: I would love to have it visually represented on the map itself. For example looking at Cloud Cap's GCS they show you the warning for terrain height in the form of a red coloration on hills and mountains. Having this greatly improves the user experience and allows for much better interaction with the mission you are flying.

Google's api has some great features currently that I think could be implemented, for example the ability to view the map in a tilt format. Apart from terrain map improvements the GCS could do well with a GUI makeover, something that I am willing to help with coming from the side of a web developer :) .

gautierhattenberger, 2015-05-10:
The GCS is pretty old (designed in 2007) and is mostly implementing all the map handling by map. And not really in a nice way most of the time. It might be a lot of work to implement "big" features, and it is written in Ocaml (which is usually not very well known...). Of course, if you're willing to give a try, you're more than welcome :)

Are you flying in a mountainous area ? If the altitude over ground really matters for you, you can test http://wiki.paparazziuav.org/wiki/GCS#Altitude_graph_widget (less convenient to use than what you suggest, but it can help)

jroboticsgroup, 2015-05-10:
Is there a way to totally replace the GCS then? like with QGroundstation

gautierhattenberger, 2015-05-10:
there is a very support of mavlink made by TUDelft guys, not sure what is the status of this. But using it, you'll loose one of the main benefit I think of Paparazzi, which is the flight plan scripting language of the flight plan.

gautierhattenberger, 2015-05-10:
by the way, it seems that our script to extract google map API number is not working any more. Can you give me the content of the file conf/maps.xml ?

gautierhattenberger, 2015-05-10:
and can you report if paparazzi/paparazzi@e762fb758938222049234ca51d9dbcdee074ca4e helps or not ? (in google_map_version branch)

jroboticsgroup, 2015-05-11:
I have this in maps.xml

<!DOCTYPE maps SYSTEM "maps.dtd">

<maps google_version="129"/>

jroboticsgroup, 2015-05-11:
where should i put paparazzi/paparazzi@e762fb7
<https://github.com/paparazzi/paparazzi/commit/e762fb758938222049234ca51d9dbcdee074ca4e>
?

gautierhattenberger, 2015-05-11:
You need to update from git and checkout this branch. Assuming the name of the remote repository of paparazzi is `origin` and you don't have local modifications (or only in your own files), something like this should work:
```
git fetch origin
git checkout -b google_map_version origin/google_map_version
```
Now you should be on the new branch. Type `make` to rebuild the project and test. When you are done `git checkout <your previous branch>`. Use `git status` to know on which branch you are and if you have modifications.

gautierhattenberger, 2015-05-14:
@jroboticsgroup any update on this issue ?

jroboticsgroup, 2015-05-14:
I have been busy, will do the changes later

pesimov, 2015-05-18:
Hi, 

I wanted to check this too. Git status is: 
On branch google_map_version
Your branch is up-to-date with 'origin/google_map_version'.

Unfortunately, the maps were not updated. 

I assume in about a week's time I could also look into it..

gautierhattenberger, 2015-05-18:
on this branch, when you type `make` at the project root, do you have something like
```
-----------------------------------------------
DOWNLOAD: google maps version code
Could not download google maps version code
-----------------------------------------------
```
or
```
-----------------------------------------------
DOWNLOAD: google maps version code
Updated google maps version to 171
-----------------------------------------------
```
?

pesimov, 2015-05-18:
Hi, 
Thanks for reminding me to run 'make.' 

It does show:

-----------------------------------------------
DOWNLOAD: google maps version code
Updated google maps version to 171
-----------------------------------------------

Furthermore, the maps load correctly now - I had missed running 'make' earlier. 

Thanks

gautierhattenberger, 2015-05-18:
closing issue with 909239b986c998dbadc9808ad93a714a311e8b46

closed 2015-05-18 by gautierhattenberger

jroboticsgroup, 2015-05-18:
Dont close as yet, havent been able to try the last fix due to being busy

gautierhattenberger, 2015-05-18:
I could confirm the fix for three people, so it should be fine. Don't worry, I'll reopen it if you report an error.

jroboticsgroup, 2015-05-18:
great great

podhrmic, 2015-05-20:
Would be possible to backport this to v.5.4?

jroboticsgroup, 2015-05-20:
Can I get detailed instruction on how to do the git update? still new to
paparazzi

gautierhattenberger, 2015-05-20:
@podhrmic backported to v5.4

@jroboticsgroup you can look my previous message to update. If you are on master branch without modification `git pull origin/master` should be enough.
If you're not confident, post the output of the command `git status` and `git remote` it will be easier to guide you.

jroboticsgroup, 2015-05-25:
I am getting this error while trying to make paparazzi

../../conf/Makefile.arm-embedded-toolchain:29: *** Error: arm-none-eabi-gcc
cross-compiler not found! Recommended toolchain is
https://launchpad.net/gcc-arm-embedded.  Stop.
make: *** [ext] Error 2

whats the most straightforward way to make this?

gautierhattenberger, 2015-05-25:
If you are using a recent ubuntu or debian, you can just do `sudo apt-get install gcc-arm-none-eabi`
Otherwise, you can try one of the following methods: http://wiki.paparazziuav.org/wiki/Installation/Linux#ARM_embedded_toolchain