paparazzi/paparazzi #1180: ARdrone2 IMU_Lost often

OpenUAS, 2015-05-15:
Using Paparazzi version v5.5_devel-586-g27baff9 ARdrone2 IMU_Lost often. Using ardrone2_raw version v5.2 no the issue. Same hardware, multiple tests.

OpenUAS, 2015-05-15:
Assigne to self, since issue now under investigation. Will add more sensible info and see I I can pinpoint the cause. It keeps raining Ardrone2's, even when the sun shines ;-)

flixr, 2015-07-07:
Any updates here?

gautierhattenberger, 2015-07-14:
It is not clear if we can really do something in about this issue in the end... It looks like a hardware flaw of the ARDrone2. Unless we have more information from Parrot, I would like to close this issue and add a warning about this known issue in the wiki.

flixr, 2015-07-14:
I'm at least taking it out of the milestone until we have more information to act on....

OpenUAS, 2015-07-15:
With the new fix of non blocking did not experience it on multiple different AR-Drones 2. The issue with testing was there where multiple other PPRZ issue keeping from having decent test flights. So closed for now, and will reopen if experienced again. @gautierhattenberger , see above Same Hardware, other PPRZ version, so yes we can do something.

closed 2015-07-15 by OpenUAS