paparazzi/paparazzi #1183: NPS Upgrade

podhrmic, 2015-05-21:
**Motivation:** Current NPS simulation works sufficiently well for many applications (FW and VTOL), however it is hardcoded to simulate gyros, accelerometer and other sensors. In case the user wants to simulate an external INS, it is not possible at the moment. Imagine I am using some external INS and don't really care about the IMU/AHRS subsystems.

**Objective:** Upgrade NPS so the sensor are simulated only when they are used in the airframe config file (i.e. Aspirin IMU is used). Otherwise some *transparent* subsystems are used. Add possibility to simulate an external INS.

### Tasks:
- [ ] Enhance NPS so the sensor functions are not hardcoded, but used only if needed
- [ ] Add option to simulate external INS (so no IMU is needed, some *transparent* subsystems might be needed in such case)
- [ ] Consider refactoring NPS using C++ and Boost library (instead of Glib), to make the code more flexible.

The last point (C++ and Boost) adds yet another dependency to Paparazzi (you would need Boost on your computer to run it), but it would make the code cleaner, more flexibly and probably easier to modify and expand.

What are your thoughts on this?

podhrmic, 2015-05-21:
@flixr @gautierhattenberger ?

gautierhattenberger, 2015-05-23:
NPS has been done to allow this kind of things (changing the sensor model), but nobody really used it in the end so it is not so easy to do since config files are not really adapted.
About using more C++ things, avoiding a big lib like boost would be good, we already have a lot of dependencies.
Also, this is a lot of work. Do you have the resources to tackle this ?

podhrmic, 2015-05-25:
I sort of need to make the first two tasks anyway, so that is not an issue. I am not sure about the refactoring - short answer is maybe.... Maybe good idea would be to look at other projects, I know that at least APM / ardupilot has some kind of SITL...

podhrmic, 2015-08-25:
Update - the current NPS actually allows the external INS to be used in a reasonably manner, so I marked it as resolved. As far as hardcoding sensors - from user experience it provides all what I needed at this point, so I marked it as resolved too.

Only question is whether to refactor it into C++ which would be nice, but at this point probably not very important.

podhrmic, 2015-08-25:
@flixr shall we close this issue?

flixr, 2015-08-25:
I'm fine with closing it...

closed 2015-08-25 by podhrmic