paparazzi/paparazzi #1184: HITL Upgrade

podhrmic, 2015-05-21:
**Motivation:** Current HITL subsystems is obsolete and not-functional. Last time it was used as is was probably sometime before 2012. HITL is a great way of testing the code, and having HITL functionality in Paparazzi by default would allow developers to test their code on real hardware in simulated flight, which adds another level of safety before new code is committed. Also users could use it for tuning control loops and simulate missions on actual hardware.

The autopilot can be connected over serial or USB port. That means we can either simulate whole packets (in case of an external AHRS or INS, when the autopilot only parses attitude and position information from an external source), or we simulate data coming from other interfaces too (SPI->gyro, i2c->baro etc). The first option is definitely closer to real flight because no code change is needed (simulation<->flight), the latter option is more thorough (and more complex) and allows the users to test paparazzi AHRS algorithms on actual hardware.

### Tasks:
I would imagine working in the following order:
- [ ] 1. remove the obsolete hitl files
- [ ] 2. update the flightgear interface to the newest version (minor change in network protocol)
- [ ] 3. develop brand new HITL program (either in C++ & Boost, or Python)
- [ ] 4. integrate HITL into paparazzi build process and with NPS

The motivation behind the last point is, that NPS already takes care of the simulation part (FDM of the aircraft). Having FDM again in HITL seems to be just a duplication. I guess in a perfect world, it would be possible to run NPS with HITL option, in which case it would receive external inputs from the autopilot and send proper data back. 

At least in the initial phase, I would focus on HITL where we assume external INS (see above the differences), so the onboard AHRS/INS subsystems wouldn't be used.

Any thoughts on this?

podhrmic, 2015-05-21:
@flixr @gautierhattenberger ?

gautierhattenberger, 2015-05-23:
Fixing HITL sim is probably a good idea. Same comment than #1183 about adding yet another big dependency like boost. I haven't used C++ for a long time, so I might be wrong about the amount of time to do this job.
You described the problem pretty well, so there is not much to add.

podhrmic, 2015-05-25:
The motivation for boost is that it will cut significantly the development time. Probably even better would be using Python, but that would require making Python bindings for JSBsim - the existing ones seem to be outdated unfortunately: http://sourceforge.net/projects/jsbsim-python/?source=typ_redirect

Adding boost should be as simple as installing one package on Ubuntu, don't know about other systems (Mac OS? Android?)

podhrmic, 2015-08-25:
A quick update - I have a working HITL in C++ (it uses Boost) and just as a demo I would like to integrate it into the Paparazzi build process (so you can compile the target the same way as NPS). I am looking into using it as another target under *fixedwing* firmware, but not sure if it is the best option.

flixr, 2015-08-25:
Ah, cool... do you have that pushed somewhere already?
Also what parts of boost are you actually using? If it's only stuff that is available in C++11 then I would strongly advocate to ditch boost and use C++11!

podhrmic, 2015-08-25:
Not pushed it yet, trying to figure out the integration with Paparazzi center first (so you can compile it with your airframe). 

I am using:
```
#include <boost/asio/serial_port.hpp>
#include <boost/asio.hpp>
#include <boost/thread.hpp>
#include <boost/date_time.hpp>

#include <boost/bind.hpp>
#include <boost/shared_ptr.hpp>
#include <boost/enable_shared_from_this.hpp>
#include <boost/asio.hpp>
```

And yes, it is C++11 (boost 1.54 I believe)

flixr, 2015-08-25:
What I mean is that using C++11 features you can probably replace most (all?) of your usages of boost by things now in the standard library, e.g. thread, shared/smart pointers, bind...

podhrmic, 2015-09-17:
please see #1360 for an actual commit

flixr, 2016-12-17:
Closed with #1831

closed 2016-12-17 by flixr