paparazzi/paparazzi #1186: paparazzi build failure

glenfordwilliams, 2015-05-24:
cant seem to get paparazzi to compile under arch linux any ideas?

Warning 3: deprecated: String.create
Use Bytes.create instead.
OC pc_control_panel.ml
File "pc_control_panel.ml", line 119, characters 29-53:
Warning 3: deprecated: String.set
Use Bytes.set instead.
File "pc_control_panel.ml", line 129, characters 35-47:
Warning 3: deprecated: String.set
Use Bytes.set instead.
OC pc_aircraft.ml
OC paparazzicenter.ml
File "paparazzicenter.ml", line 154, characters 15-28:
Warning 3: deprecated: String.create
Use Bytes.create instead.
OL paparazzicenter
make -C sw/ground_segment/misc
CC davis2ivy
CC kestrel2ivy
CC natnet2ivy
CC video_synchronizer
make -C sw/logalizer
DEPEND .depend
OC log_file.ml
OC play_core.ml
OC play.ml
OL play
OC plotter.ml
OL plotter
GLADE gtk_export.ml
OC gtk_export.ml
OCI export.mli
OC export.ml
OC plot.ml
OL plot
OC sd2log.ml
File "sd2log.ml", line 88, characters 14-27:
Warning 3: deprecated: String.create
Use Bytes.create instead.
OL sd2log
CC plotprofile
CC openlog2tlm
make -C sw/ground_segment/cockpit
/bin/sh: bc: command not found
DEPEND .depend
/bin/sh: bc: command not found
GLADE gtk_setting_time.ml
OC gtk_setting_time.ml
GLADE gtk_strip.ml
OC gtk_strip.ml
OCI horizon.mli
OC horizon.ml
OCI strip.mli
OC strip.ml
GLADE gtk_save_settings.ml
OC gtk_save_settings.ml
OCI saveSettings.mli
OC saveSettings.ml
OCI page_settings.mli
OC page_settings.ml
OCI pages.mli
OC pages.ml
OCI speech.mli
OC speech.ml
OCI plugin.mli
OC plugin.ml
OCI sectors.mli
OC sectors.ml
OCI map2d.mli
OC map2d.ml
OCI editFP.mli
OC editFP.ml
OCI live.mli
OC live.ml
OC particules.ml
OCI papgets.mli
OC papgets.ml
OC gcs.ml
OL gcs
make -C sw/ground_segment/multimon
CC hdlc.c
CC pprz.c
1CC demod_afsk48p.c
1CC demod_display.c
CC unixinput.c
CC xdisplay.c
LD multimon
CC pprzlib.c
CC demod_afsk12.c
CC demodml.c
CC mkcostab.c
LD mkcostab
EXEC mkcostab
CC costabi.c
CC gen_hdlc.c
CC ml_hdlc.c
OC demod.ml
OC hdlc.mli
OC hdlc.ml
OLD multimon.cma
demod_afsk12.o: In function `hweight32':
demod_afsk12.c:(.text+0x400): multiple definition of `hweight32'
pprzlib.o:pprzlib.c:(.text+0x0): first defined here
demod_afsk12.o: In function `hweight16':
demod_afsk12.c:(.text+0x460): multiple definition of `hweight16'
pprzlib.o:pprzlib.c:(.text+0x60): first defined here
demod_afsk12.o: In function `hweight8':
demod_afsk12.c:(.text+0x4b0): multiple definition of `hweight8'
pprzlib.o:pprzlib.c:(.text+0xb0): first defined here
demod_afsk12.o: In function `gcd':
demod_afsk12.c:(.text+0x4e0): multiple definition of `gcd'
pprzlib.o:pprzlib.c:(.text+0xe0): first defined here
demod_afsk12.o: In function `lcm':
demod_afsk12.c:(.text+0x520): multiple definition of `lcm'
pprzlib.o:pprzlib.c:(.text+0x120): first defined here
demod_afsk12.o: In function `mac':
demod_afsk12.c:(.text+0x570): multiple definition of `mac'
pprzlib.o:pprzlib.c:(.text+0x170): first defined here
demod_afsk12.o: In function `fsqr':
demod_afsk12.c:(.text+0x5a0): multiple definition of `fsqr'
pprzlib.o:pprzlib.c:(.text+0x1a0): first defined here
collect2: error: ld returned 1 exit status
Makefile:75: recipe for target 'multimon.cma' failed
make[1]: *** [multimon.cma] Error 2
Makefile:142: recipe for target 'multimon' failed
make: *** [multimon] Error 2

gautierhattenberger, 2015-05-25:
Please use the mailing list or the Gitter chat room (http://wiki.paparazziuav.org/wiki/Contact) for this kind of question.
You should also add the version of paparazzi (run `./paparazzi_version`) and the version of gcc (`gcc --version`).

closed 2015-05-25 by gautierhattenberger

flixr, 2015-07-31:
Should be fixed with https://github.com/paparazzi/paparazzi/pull/1276