paparazzi/paparazzi #1199: Add rotorcraft support for <Follow> flight plan block

hooperfly, 2015-06-04:
The `follow` flight plan block is only supported for `fixed wing` UAS at the moment. It would be nice to have this capability within the `rotorcraft` domain as well. Some design decisions related to how this feature functions across UAS types (i.e. `fixed wing`, `rotorcraft`, `vtol`, etc....) is open for discussion.

Wiki reference: http://wiki.paparazziuav.org/wiki/Flight_Plans#Follow

gautierhattenberger, 2015-06-04:
Main issue is that reference point of rotorcraft are (usually) different, so you can't compare their local position easily. It means that we need to compute a local position from a global one (ecef, lla, utm ?) each time a traffic_info message is received.
Currently this message is sending utm, which is fine for fixedwing, but not really for rotorcraft.

hooperfly, 2015-06-04:
Can you explain more why UTM is ok for fixedwing vs. rotorcraft? With respect to reference point, I would assume that all aircraft taking part in multi-aircraft exercise would share a common reference point in each of their flight plans. Is this not the case usually? Does reference point mean something other than a (x,y,z)->(0,0,0) reference value?

flixr, 2015-06-08:
@hooperfly in the fixedwingfirmware the navigation coordinate system is UTM, in rotorcraft firmware it is LTP

hooperfly, 2015-06-08:
@flixr thanks. I've been reading some archive UTM/LTP fixed/rotor discussion to become more familiar with the issues: https://gitter.im/paparazzi/discuss/archives/2015/01/05

gautierhattenberger, 2021-03-25:
there is a follow nav module for rotorcraft if needed http://docs.paparazziuav.org/latest/module__follow.html

closed 2021-03-25 by gautierhattenberger