paparazzi/paparazzi #1201: HOVER and NAV problem when using YAW command

agressiva, 2015-06-08:
When using YAW command in HOVER or NAV the rotorcraft start wobble.

I tested all combinations of AHRS and Stabilization and the problem only occurs when using stab_int_quat or stab_float_quat

in this video is possible to see the problem:
look at 3:00 to see how crazy the problem is.
https://www.youtube.com/watch?v=wmKrjwOyuAQ

agressiva, 2015-06-08:
|                |                  Stabilization                    |
+----------------|---------------------------------------------------|
|     AHRS       | int_quat   | float_quat | int_euler | float_euler |
| int_cmpl_quat  |   wobble   |little wob  |     OK    |     Ok      |
| float_cmpl     |   wobble   |little wob  |     OK    |     OK      |
| int_cmpl_euler |            |     -      |     -     |      -      |
| float_mlkf     |   wobble   |     -      |     OK    |      -      |

agressiva, 2015-06-08:
logs that sshow the problem
https://www.dropbox.com/home/Public/Paparazzi%20wobbly

flixr, 2015-06-08:
That link is only valid for you...

flixr, 2015-06-08:
might be related to #161?

agressiva, 2015-06-08:
Felix, i read the #161 and i think its related.
some time ago (when i started to fly ppz quadrotor) i do some fast maneuver (coordinated turns, funnel pitch etc) and some time after fast yaw changes the attitude work very bad for some time.

This week i do the same maneuver using euler and not see any disturbance in attitude.

so, i have a question:  way the heading use to rotate the matrix is the desired not the real heading ?

flixr, 2015-06-09:
because you want to compute the final setpoint and currently it simply treats all attitude axes as equivalent as @dewagter noted...

flixr, 2015-07-25:
To clarify: you only have this issue if you give a "full" yaw command and when you yaw more slowly everything is fine, correct?
That would also mean you should adjust the yaw part attitude reference model (lower it to something your airframe can actually achieve).

agressiva, 2015-07-25:
Felix, the first time i see this problem the copter are flying in NAV.
It occurred when i changed from STANDBY to rectangle_survey and the copter start to climb and changing the yaw at same time.

the copter crashed two times in this situation. Its very dangerous because the copter goes do ground with full throttle.

The 3th time it occur the copter are in rectangle survey and i start another polysurvey with plus 90 deg.

I am not confident with quaternion stabilize. I will not use it anymore.

BTW, i am doing the tests with full YAW because its the way i found to simulate the problem but some time the problem occur even with low yaw rate. All depend of the external disturbance when YAW is applied.

OpenUAS, 2015-11-20:
I recall I've seen this issue also. 
Action: Will retest if #161 is related and fixes it.

dewagter, 2015-11-23:
Dear Eduardo,

Sorry to hear about the crashes. The proper handling of saturations is very
complex in multicopters (as it is not obvious when a certain motor
saturates, which command is "most" responsible and should be reduced). This
is being investigated.

In the mean time there might be a workaround: namely to significantly
reduce the max yaw rate and yaw acceleration in the reference model. This
way you can safely command 180 deg turns, but the command (reference) will
be a slow controlled turn.

-Christophe

OpenUAS, 2015-12-10:
So a "fix" and even investigation is ongoing, great. If one could add and example airframe and which (all) value to change as example, then add to this issue. Then defect(so we can somewhere stuff it on the wiki) we could close the issue IMHO

agressiva, 2015-12-10:
Well, i stopped the tests because my quad copter crashed on the last one.

hooperfly, 2015-12-10:
@agressiva I think you need a HooperFly for those kind of tests. :)

agressiva, 2015-12-11:
for sure ...

podhrmic, 2016-07-06:
Is this still an issue? Maybe can be closed because of feedback timeout or at least moved to later release?

agressiva, 2016-07-06:
the problem persist.

hooperfly, 2016-07-07:
@agressiva +1. I would recommend we keep this open for now, it's still an issue.

gautierhattenberger, 2023-12-04:
Not relevant anymore

closed 2023-12-04 by gautierhattenberger