paparazzi/paparazzi #1202: HOVER and NAV wobble

agressiva, 2015-06-08:
When using stabilization int_euler or float_euler the hover/navigation start wobble 

here the log that show the problem.
https://www.dropbox.com/sh/abo1p42zsfx1k8e/AABjgGWHmkGspIFlX8tJey19a?dl=0
I am using the latest code whitout modification.
AHRS = int_cmpl_quat
STAB =  int_euler
INS = hff

flixr, 2015-06-08:
duplicate of #1201 ?

flixr, 2015-06-08:
And you only have the problem with euler stabilization?
Why use euler in the first place (instead of the quaternion version)?

Also the log is from Quad680 with float_euler stabilization and float_mlkf ahrs...

agressiva, 2015-06-08:
Hi Felix, i ever used quaternion  but last week i do some flightplans using YAW command and found a problem with this stabilization.

Every time i use the YAW the quad do some strange and dangerous maneuver so i changed to euler and i get another problems.

flixr, 2015-06-08:
Did you test #1170 (which should bring some improvements)?

flixr, 2015-06-08:
From the log it looks like you request a higher yaw rate than the quad can manage...

agressiva, 2015-06-08:
Felix, i tested using #1170 without any result.

about yaw rate, yes ... quad not support that acceleration.
I do some adjusts in reference to do the tests.

OpenUAS, 2015-06-09:
The hint by @flixr about to high a to high yaw rate is also in our experience cause for "Wobble". Let us know via this issue it adjusting the thing mentioned fix your issue and add comment to this issue please, that would be great.

agressiva, 2015-07-16:
I do some more tests with another quadcopter and it worked fine.
The problem is related to mine hardware.

agressiva, 2015-07-16:
tested with stabilization int and float. 
work perfect.

closed 2015-07-16 by agressiva