paparazzi/paparazzi #1208: Flashing using BMP  [upload] Segmentation fault

OpenUAS, 2015-06-16:
Using PPRZ v5.5_devel-628-gd078f7f  Flashing a regular Lisa/M v2 using a Black Magic Probe (BMP) with firmware blackmagic-20141017.bin gives during [upload] a Segmentation fault

Paparazzi version v5.4.2_stable-0-g21e8716 works well, exact same hardware.

...
No. Att Driver
 1      STM32, High density.
make[1]: *** [upload] Segmentation fault (core dumped)
...

flixr, 2015-06-16:
This doesn't seem to be a paparazzi problem, you should open an issue with the black magic probe....

podhrmic, 2015-06-16:
does BMP use llibopencm3? Maybe it can be related to our recent upgrade of libopencm3 version?

flixr, 2015-06-16:
AFAIK, the BMP does use libopencm3, but since we don't build the bmp firmware in paparazzi this is unrelated to our libopencm3 update.

flixr, 2015-06-16:
Also I can upload just fine using paparazzi version v5.5_devel-628-gd078f7f, Black Magic Probe (Firmware 1.5, build 20141016), GNU gdb (GNU Tools for ARM Embedded Processors) 7.8.0.20150304-cvs

flixr, 2015-06-23:
@OpenUAS as I cannot reproduce this, you will have to dig a bit deeper yourself...

esden, 2015-06-24:
This looks like arm-none-eabi-gdb crashing. So I would try to investigate if this is related to the version of your gdb vs BMP firmware...

Did you try to open the core dump and see what crashes?

gautierhattenberger, 2015-06-26:
Some more info about a recent issue.
While flashing a (working) code (not paparazzi actually) with a BMP and the latest firmware, the program (running on F4) randomly crashes (during peripherals init). Doing the same with an old BMP firmware, the same code is working as expected.
The old working firmware was https://github.com/blacksphere/blackmagic/commit/e0a8ce5a887cb31f38077cd22271fd894070e0e9
We didn't investigate further, but there is probably a serious issue between this point and the latest version.

flixr, 2015-06-30:
@OpenUAS @podhrmic could you plz test setting the debug flag for compiling back to -g?
```
make AIRCRAFT=foo clean_ac ap.upload DEBUG_FLAGS=-g FLASH_MODE=SWD
```
as #1164 is the only change between v5.4 and current master that I can think of which might impact flashing.
If it still fails, could you please try flashing a firmware built with master with v5.4?
And try flashing "manually", so something like
```
arm-none-eabi-gdb
target extended-remote /dev/ttyACM0
monitor version
monitor swdp_scan
attach 1
load
kill
```
to see when it segfaults?

podhrmic, 2015-07-01:
With ``DEBUG_FLAGS=-g`` it uploads no problem! If I remove the ``DEBUG_FLAGS`` option, it segfaults during upload.

podhrmic, 2015-07-01:
Interestingly "manual" flashing works with either DEBUG_FLAG.
Also, if I run the black_magic_script manually (step by step), it also works. Just when I do it with the make command, it fails

flixr, 2015-07-01:
Huh? that is really weird...
at which point does it fail? After the load?

podhrmic, 2015-07-01:
it crashes during the load:
```
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>

Target voltage: 3.3V
Available Targets:
No. Att Driver
 1      STM32F4xx
make[1]: *** [upload] Segmentation fault (core dumped)
```

but not if I enter the same commands manually in gdb

flixr, 2015-07-01:
What gdb version are you using?
I guess it then still crashes if you remove the kill from the script?
And just to make absolutely sure that it is a gdb/bmp problem, it also crashes if you run gdb in batch mode with the script from the commandline, right?
```
arm-none-eabi-gdb -batch -ex 'target extended-remote /dev/ttyACM0' -x sw/tools/flash_scripts/bmp_jtag_flash.scr 
```
Also, do you have anything in your `.gdbinit`? (as that is also executed in batch mode it seems..)

podhrmic, 2015-07-01:
OK, let me rephrase it.

### Case 1 ###
1. Complie AC code from terminal with the DEBUG_FLAGS ``make AIRCRAFT=foo clean_ac ap.upload DEBUG_FLAGS=-g FLASH_MODE=SWD`` => it uploads just fine
2. Uploading the same ``ap.elf``file manually, i.e. running these steps
```
$ arm-none-eabi-gdb ap.elf
target extended-remote /dev/ttyACM0
monitor version
monitor swdp_scan
attach 1
load
kill
```
works just fine too.

### Case 2 ###
1. Complie AC code from terminal *without* the DEBUG_FLAGS ``make AIRCRAFT=foo clean_ac ap.upload  FLASH_MODE=SWD`` => it crashes after attaching the target
```
Using Black Magic Probe with SWD on BMP_PORT /dev/ttyACM0
Using GDB = arm-none-eabi-gdb
  BMP	/home/fwmav/Paparazzi/master/can_test/paparazzi/var/aircrafts/Quad_LisaMX/ap/ap.elf
Black Magic Probe (Firmware 1.5-00279-gfd9eef8, build 20141114)
Copyright (C) 2011  Black Sphere Technologies Ltd.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>

Target voltage: 3.3V
Available Targets:
No. Att Driver
 1      STM32F4xx
make[1]: *** [upload] Segmentation fault (core dumped)
make: *** [ap.upload] Error 2
```
2. Running the same steps manually (the same code):
```
$ arm-none-eabi-gdb ap.elf
target extended-remote /dev/ttyACM0
monitor version
monitor swdp_scan
attach 1
load
kill
```
it segfaults after ``attach1`` command.
3. But if I start gdb without specifying the file to be loaded and use the following command sequence:
```
$  arm-none-eabi-gdb
target extended-remote /dev/ttyACM0
monitor version
monitor swdp_scan
attach 1
load ap.elf
kill
```
then it works just fine...

Btw I am using this version of gdb:
```
$ arm-none-eabi-gdb --version
GNU gdb (GNU Tools for ARM Embedded Processors) 7.4.1.20130613-cvs
Copyright (C) 2012 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "--host=i686-linux-gnu --target=arm-none-eabi".
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
```

flixr, 2015-07-01:
Could you please also try with a more recent gdb version?
E.g. I couldn't reproduce this with gdb 7.8.0.20150604-cvs that is included in gcc-arm-none-eabi 4.9.3.2015q2-1trusty1 from [Terry.Guo] (https://launchpad.net/~terry.guo/+archive/ubuntu/gcc-arm-embedded)

podhrmic, 2015-07-01:
works with newer (7.8)GDB version without a problem.
It is possible that I had the Ubuntu package, not the ARM one, yet it worked without problems until now (so I haven't noticed).
So the mitigation is to upgrade your arm-none-eabi version and use the correct package.

flixr, 2015-07-01:
Yep, closing this then, as it is not a paparazzi bug.
Solution: use a recent gdb version, 7.8.0.20150604-cvs that is included in gcc-arm-none-eabi 4.9.3.2015q2-1trusty1 from [gcc-arm-embedded on lauchpad] (https://launchpad.net/~terry.guo/+archive/ubuntu/gcc-arm-embedded)

closed 2015-07-01 by flixr