paparazzi/paparazzi #1209: add copy AC menu item to paparazzi center

flixr, 2015-06-18:
I think it would be nice to add a "A/C->Copy" menu item to the Paparazzi Center menu additionally to "A/C->New".
Often one starts with one of the existing airframes anyway and goes on to adapt it.
So it would be desirable to have a "copy" option that creates a new A/C entry in conf.xml with the same initial xml files as the currently selected one. Preferably there would also be a dialog to ask if the airframe file of the chosen aircraft should be copied and saved under a new name.

TobiasMue, 2015-06-20:
great idea!

OpenUAS, 2015-07-15:
Only if copy is copy it will be handy IMHO, if not the Airframe XML is copied it is more like an alias an will give much confusion(based in hands on with +100 different students using Paparazzi)

OpenUAS, 2015-11-05:
Squeezing it in for the overnext release because of positive response and usefulness

commit 0b1a15d1 (2015-11-11): Merge pull request #1406 from paparazzi/ac_copy

[paparazzi center] add copy A/C menu entry

On copy it will ask for a new aircraft name (assign the next free id) and offer to copy the airframe file to a new one with different name. If the airframe file name is not changed, it will continue to use the same one.
Closes #1209
closed 2015-11-11 by flixr