paparazzi/paparazzi #1210: Use 8 bit I2C address scheme for Linux drivers

martinmm, 2015-06-19:
The imu_mpu60x0_i2c subsystem does not work with STM32 targets. The address gets shifted by one bit in in the i2c init call

https://github.com/paparazzi/paparazzi/blob/master/sw/airborne/subsystems/imu/imu_mpu60x0_i2c.c#L74

as in other Linux drivers, e.g. Bebop

https://github.com/paparazzi/paparazzi/blob/15f8c182544d6197b21f4da2c88c5b177453abc3/sw/airborne/subsystems/imu/imu_bebop.c#L82

Seems the Paparazzi 8 bit I2C address convention collides with the 7 bit convention in Linux.

ioctl(file, I2C_SLAVE, t->slave_addr);

https://github.com/paparazzi/paparazzi/blob/master/sw/airborne/arch/linux/mcu_periph/i2c_arch.c#L56

ioctl(file, I2C_SLAVE, long addr)
  Change slave address. The address is passed in the 7 lower bits of the
  argument

http://git.kernel.org/cgit/linux/kernel/git/torvalds/linux.git/tree/Documentation/i2c/dev-interface#n103

Wouldn't it be better to do this bit shift once in the Linux i2c_arch.c rather than in the individual (Linux-only) drivers to keep the code clean and consistent or do I miss something?

flixr, 2015-06-21:
Yes, we should definitely be consistent here!
The question is if we want to use the 7-bit or 8-bit convention? I guess keeping the 8-bit convention in our mcu_periph API is the least intrusive as you proposed...
Could you make a patch/pull request?

martinmm, 2015-06-21:
Would vote for 8 bit and will create a PR

PR #1211 (2015-06-22): [linux] use 8 bit I2C address scheme for all drivers
fixes #1210
needs to be tested as I do not have any Linux target (Bebop, BBB)
closed 2015-06-22 by flixr