paparazzi/paparazzi #1212: input2ivy can't send all messages

flixr, 2015-06-23:
input2ivy is partially broken since #1138 since now message field types have to match.
This is not the case in input2ivy where all fields are simply ints (which is a limitation anyway...).
So it fails for any message where the fields are not ints with something like:
`Scanf.Scan_failure("format read '%s' does not match specification '%d'")` errors

Steps to reproduce:
Run input2ivy with a config file that sends e.g. the DL_SETTING or JUMP_TO_BLOCK message
e.g run `sw/ground_segment/joystick/input2ivy -ac bebop logitech_f710.xml`

flixr, 2015-06-23:
Thx @gautierhattenberger while this seems to have fixed the immediate issue, you can still not send any float values via input2ivy (e.g. in DL_SETTING).

gautierhattenberger, 2015-06-23:
Crap, I didn't realized that my tests were all using integer...

gautierhattenberger, 2015-06-23:
is it better now ?

flixr, 2015-06-24:
A bit nicer, but `eval_expr` can still only return ints and will `failwith "eval_expr: float"` if you have a float as value in a message field.

gautierhattenberger, 2015-06-24:
This was already the case before I think. The easy way is to make an int from a float when in occurs, but it is not really satisfactory. We can do the opposite: always return float and truncate afterwards to int if needed. Correctly handling both is of course possible, but a bit annoying...
In which case do you really need a float ?

flixr, 2015-06-24:
Yes, that was already the case before, so the really vital part is fixed.
And I don't have an immediate need to send any float values, but in general it should be possible (e.g. if you want to change a setting that is really a float).
Shall I open a new issue for that, or do we track this in here?

gautierhattenberger, 2015-06-24:
Better close this one and open a new one. But this will actually a lot of trouble to support this from my first investigation.

commit 8270d108 (2015-06-23): [input2ivy] build message from string and not values

close #1212
closed 2015-06-24 by gautierhattenberger