paparazzi/paparazzi #1213: WEAK functions cannot be properly redefined

podhrmic, 2015-06-23:
**How to reproduce the bug?**

1. Paparazzi master, select A/C Twinjet
2. in https://github.com/paparazzi/paparazzi/blob/master/sw/airborne/subsystems/imu/imu_analog.h#L56 comment out lines 56 to 61 
3. Compile - you will see the following error:
```
subsystems/imu.c:230:20: error: 'IMU_ACCEL_X_SENS_NUM' undeclared (first use in this function)
subsystems/imu.c:230:20: note: each undeclared identifier is reported only once for each function it appears in
subsystems/imu.c:230:44: error: 'IMU_ACCEL_X_SENS_DEN' undeclared (first use in this function)
subsystems/imu.c:232:20: error: 'IMU_ACCEL_Y_SENS_NUM' undeclared (first use in this function)
subsystems/imu.c:232:44: error: 'IMU_ACCEL_Y_SENS_DEN' undeclared (first use in this function)
subsystems/imu.c:234:20: error: 'IMU_ACCEL_Z_SENS_NUM' undeclared (first use in this function)
subsystems/imu.c:234:44: error: 'IMU_ACCEL_Z_SENS_DEN' undeclared (first use in this function)
```

### Summary ###
```imu_analog``` is redefining the default ```imu_scale_accel``` function (here https://github.com/paparazzi/paparazzi/blob/master/sw/airborne/subsystems/imu.c#L226 ), with an empty function (https://github.com/paparazzi/paparazzi/blob/master/sw/airborne/subsystems/imu/imu_analog.c#L118) so no ACCEL_*_SENS values should be necessary. 

However the weak function isn't replaced with our implementation as it should, so the original function https://github.com/paparazzi/paparazzi/blob/master/sw/airborne/subsystems/imu.c#L226 is still called.

I noticed identical behaviour for ```imu_scale_gyro``` and for other imus (imu_

flixr, 2015-06-23:
This is expected (while not necessarily desired).
If you have a weak function, it is still always compiled. But if you link and you provide a non-weak function, then this is used/linked in instead.
In this context it means that in order to compile the weak (default) function, you need all the defines it uses so it can compile that compilation unit.

podhrmic, 2015-06-23:
I see. So basically it means I still have to provide all necessary defines, but my own (in this case an empty function) will be called during execution, right?

Maybe in that case it would make sense to provide default defines with proper ifdef guards in imu.h?

flixr, 2015-06-23:
Correct.

The reason that I didn't put default defines in imu.h was that I wanted to get a compile error if you don't specify them (unless the particular imu implementation you use already provides some defaults).

Before doing that, I would rather split out these functions into a separate file, so they are only compiled if they are actually needed...

podhrmic, 2015-06-23:
OK, I am marking this as closed, as separating the weak functions should be another issue, and this is not really a bug.

closed 2015-06-23 by podhrmic