paparazzi/paparazzi #1233: Autopilot uses 100% cpu on ARdrone2

kevindehecker, 2015-07-10:
Using newest master, ap.elf uses 100% cpu on the ARdrone2. Also tested with 5.4.2 stable tag, which does not have the problem (around 2% according to top).

kevindehecker, 2015-07-13:
We tracked the problem to this commit e2b741be03cd4e5c5a81a88c525e7d30a9c508be

([ardrone] Make navdata multi-threaded because of blocking read)

flixr, 2015-07-13:
Yes, @fvantienen moved the reading of navdata to a separate thread (see discussion in #1117).
But I don't think that this is the "problem", rather this now enables the rest of the code to run "normally" again.
If you look at `main_event()` in main.c, you will see that this essentially runs as fast as possible, hence you have 100% cpu load while spending most of the time in this "event loop" polling for new events.

kevindehecker, 2015-07-13:
Yeah, apparently that commit fixed a long winded problem of a blocking read on the nav board, which should not block but did anyway. As a result, the ap used far less cpu. 

The main loop in the ap is now always spinning 100% by design, which makes sense with mcu architectures, but does not when an OS comes to play. It seems very clear to me that this (in case of an OS) is a bad design, but since everybody asks why here are the highlights:
-100% cpu uses more power (bad for heat and battery )
-by having a process taking up 100%, other process are affected even if that process has higher priority. (ARDrone2 does not have a RTOS, and even if it was RTOS some performance effects remain)

 Currently, I'm able to fix the problem simply by adding a usleep(1000) in the main loop (ARDrone2 flight seems unaffected).

flixr, 2015-07-13:
Of course, mainly wanted to point out that this is not a "regression" and everything was peachy in v5.4...
We all totally agree that this "event loop" is not really what you want to use if you have an (RT)OS...
There are a couple of issues/feature requests open in that direction (e.g. https://github.com/paparazzi/paparazzi/issues/243)
Any help on this is of course appreciated...

kevindehecker, 2015-07-14:
Took a quick look at #243, and did not see a quick but relevant way that would help for this specific report.

However, I'm quite happy with my quick usleep in the main loop fix, and at the least it seems much better than doing nothing.

PR #1240 (2015-07-15): [rotorcraft][linux] limit main/event loop to 1kHz
This is a kludge until we can better leverage threads and have real events.
Without this limit the event flags will constantly polled as fast as possible,
resulting on 100% cpu load on boards with an (RT)OS.
On bare metal boards this is not an issue, as you have nothing else running anyway.

Enable this for the bebop and ardrone2 by default.

Alternative to #1239 and should fix #1233
commit 4eaad2c9 (2015-07-15): Merge pull request #1240 from paparazzi/main_event_1khz_limit

[rotorcraft][linux] limit main/event loop to 1kHz

This is a kludge until we can better leverage threads and have real events.
Without this limit the event flags will constantly polled as fast as possible,
resulting on 100% cpu load on boards with an (RT)OS.
On bare metal boards this is not an issue, as you have nothing else running anyway.

Enable this for the bebop and ardrone2 by default.

Alternative to #1239 and should fix #1233
commit da0dac74 (2015-07-15): [rotorcraft][linux] limit main/event loop to 1kHz

This is a kludge until we can better leverage threads and have real events.
Without this limit the event flags will constantly polled as fast as possible,
resulting on 100% cpu load on boards with an (RT)OS.
On bare metal boards this is not an issue, as you have nothing else running anyway.

Enable this for the bebop and ardrone2 by default.

Alternative to #1239 and should fix #1233
closed 2015-07-15 by flixr