paparazzi/paparazzi #1236: Prepare flight tested airframes + select the unsoported ones

podhrmic, 2015-07-14:


flixr, 2015-07-14:
see also #888 and #974

podhrmic, 2015-07-15:
So I did a quick check of which configurations/airfames are actually compiling. A summary is here: https://docs.google.com/spreadsheets/d/1JK6Di9Z9cGQN0RycM6JxvBHufjcdzLZuKnN8s68IXP4/edit?usp=sharing

I am not sure what should be the next step - IMHO I should delete the non-working configurations and move the related airframes into some "obsolete" folder.

As for the control panels - I did some tests too (on the second page of the spreadsheet) and I think we should keep only the ``control_panel_example.xml``, the rest should go to obsolete folder as well (``airframe/TUDeldf/control_panel.xml`` and ``airframes/OpenUAS/opensuas_controL_panel_obc2014.xml``)

Let me know if you agree.

flixr, 2015-07-15:
At least TwoSeas should compile (hopefully) work again after fixing ins xsens700 in 3d41040278f724cffe9122bf37eebbc8bc6f3c71

For the personal/institution confs/airframes, I would leave it to @gautierhattenberger and @dewagter to fix/move/delete them as they see fit.

IMHO the more important part is checking/moving/fixing all the airframe files in examples that are not in any conf.
In the end all that work should be in a conf that can be tested...

podhrmic, 2015-07-16:
OK, so I did check for which aiframes are actually used in any conf file. From total 176 aiframe config files, there is 96 used and 85 unused. Please see this spreadsheet for details: https://docs.google.com/spreadsheets/d/1JK6Di9Z9cGQN0RycM6JxvBHufjcdzLZuKnN8s68IXP4/edit?usp=sharing

Please look through the list, I will make a pull request with the unused airframes being moved to the obsolete folder.

podhrmic, 2015-07-17:
@flixr @gautierhattenberger any feedback on this?

flixr, 2015-07-17:
Thanks a lot for going through all those!

As I said before, I would leave it up to the respective people to cleanup their own subfolders.
For the files in examples, I would say we move all that are not part of a conf to an `unmaintained` or `untested` folder.

OpenUAS, 2015-07-29:
Repairing, Preparing etc. some airframes

closed 2015-09-23 by flixr