paparazzi/paparazzi #1245: Problem when using remote GCS

agressiva, 2015-07-16:
i am running the SIMULATION and GCS on raspberry pi2 and
a remote GCS on mine laptop across ethernet link.

On raspberry i am using the user TESTE so the paparazzi are in home/TESTE/paparazzi.
On laptop i am using user EDUARDO so the paparazzi are in home/EDUARDO/paparazzi.

when i start the GCS on laptop i get this message:
Fatal error: exception Failure("File not found: /home/TESTE/paparazzi/var/aircrafts/Twinjet/flight_plan.xml")

as you can see, the laptop trying to use the paparazzi path of server computer and not the local path.

flixr, 2015-07-16:
The server sends the complete path in to the xml files in the CONFIG message:
```
'29752_1 ground CONFIG 202 file:///home/flixr/code/paparazzi/var/aircrafts/bebop/flight_plan.xml file:///home/flixr/code/paparazzi/conf/airframes/bebop.xml file:///home/flixr/code/paparazzi/conf/radios/dummy.xml file:///home/flixr/code/paparazzi/var/aircrafts/bebop/settings.xml red bebop'
```
Maybe we could "fix" this by inserting a placeholder for `PAPARAZZI_HOME` in the file url....

But the then you have to make sure that the files you have locally match the files on the remote host where the server is running...

The better solution for this use case might be to use the `-http` option of the server and start a http server on the remote host. Then the GCS will fetch the airframe files via http and you will surely have the correct ones...

See also #1246 and #1247

OpenUAS, 2015-07-22:
@agressiva Make a symbolic link both with same name perhaps would be an intermediate solution

flixr, 2015-07-23:
I would really suggest serving the xml files via http: https://wiki.paparazziuav.org/wiki/Server#Serving_XML_files_via_http

The question however is what we actually want to happen in a case like this?

flixr, 2015-07-25:
@gautierhattenberger what do you think?

Is it desirable to be able to use this on different hosts and users (thus different PAPARAZZI_HOME) but with the need to keep stuff in sync yourself?
The we could probably implement some placeholder for PAPARAZZI_HOME..

Or should we mark this as WONTFIX and recommend to server the files via http in cases like this?

gautierhattenberger, 2015-07-25:
Since the solution exists with http (is kind of made for this situations), I suggest to recommend it and close the issue.

flixr, 2015-07-25:
Closing, please use the http option if you want to run server and GCS on different hosts

closed 2015-07-25 by flixr