paparazzi/paparazzi #1254: Bootloader on MX not tested

OpenUAS, 2015-07-22:
AS a user I want a simple way to upload my AP software to the AP board. the newer Lisa v2 MX has an F4. THe working of flasing by USB are not tested and know to be problematic. Will dig up a board and start testing.

podhrmic, 2015-07-24:
FYI there already exists a bootloader for Lisa MX https://github.com/podhrmic/aggieair-bootloader but it still has to be merged to the main repo, since it is using older version of libopencm3 and needs to be updated. Again, something I will do after the v5.6 release.

podhrmic, 2015-07-24:
Shall we move it to milestone v5.8?

OpenUAS, 2015-07-27:
Moved for a 2nd time is not really the best thing IMHO... I've assigned it to myself so we could also fix it in the lab and a "slight" promise of Piotr ;) Note that it is more for his sake than mine, I still have a bunch of Lisa MX.

OpenUAS, 2015-07-27:
Well should I just put  a bounty on it ;)

OpenUAS, 2015-07-29:
move to https://github.com/paparazzi/luftboot/issues/8

closed 2015-07-29 by OpenUAS