paparazzi/paparazzi #1258: Automatically start/stop new log from flightplan

kevindehecker, 2015-07-23:
I would like a feature in which the log (which is important for my science) is automatically split into relevant separate files, i.e. start a new log file at take off, stop it just after landing. 

In my opinion, this could be a command called from the flight plan.

flixr, 2015-07-23:
This could be done for some of the onboard loggers, but I don't really see how it would be possible to start/stop the log created by server on the ground_segment....

agressiva, 2015-07-24:
Interesting to people that use a remote server running on raspi as i .

hooperfly, 2015-07-24:
I post process the log files and simple decompose/filter them based on whatever criteria I am interested in analyzing. Often it's simple sed/awk command on the generated log file. That way, I can tease out the relevant data. You could also use the plotter/logplotter tool to filter/decompose a log file. So maybe a post-processing tool vs. a pre-execution configuration might be more flexible? Maybe?

kevindehecker, 2015-07-24:
Post-processing: that's what I'm doing now yes. But like many, I'm not as handy in sed/awk commands...

Come to think of it, appending is much easier than splitting. Also from cmd. How about splitting logs at engine start by default? This way no flight plan signal is required. Also it smooths a regular use case for these logs, which is to debug the latest unexpected problem/crash in the field. If you have been running server for a few hours before, this is a very painful process.

flixr, 2015-07-24:
Just keep in mind that server logs all aircrafts, not just one...
So automatically splitting logs when one starts the engine is not a good default.

hooperfly, 2015-07-24:
The plotter tools already parse the log files. Enhancing them or building a new tool that facilitates the splitting/filtering of data based on various attributes(i.e.: aircraft id, message types, event types, temporal boundaries, etc...) seems like a nice way to facilitate data log transformations. Just an idea.

kevindehecker, 2015-07-25:
Currently the plotter takes ages to open of them big log files, but that idea could work!

gautierhattenberger, 2015-07-25:
As a reminder, you can use the log plotter to select a "time frame" and then export the data in CSV (http://wiki.paparazziuav.org/wiki/Plotter#Export_values_in_CSV_format). The wiki probably needs some update thought.
Also, an idea to make the slitting easier is to "generate" a message in the log that will be easy to find afterwards. I suggest to use the `Mark` button in the strip (http://wiki.paparazziuav.org/wiki/GCS#Actions, again not really well explained) that will place a blue dot on the map at your current position (which you don't care). Just search for the `MARK` message in the log and you're good.

OpenUAS, 2015-08-05:
If you fly alone, one airframe in air as intermediate solution, just stop server after a land in Paparazzi Center and restart just before takeoff. Not idea but you will get a new log

flixr, 2016-12-16:
If this is important, you could also add a new message that instructs the server to create a new log file.
You could send this message either from some ground application or from the aircraft (could be called from the flight plan).
Although this would probably only make sense if you have only one aircraft...