paparazzi/paparazzi #1263: Bebop GCS hight goes up and up

OpenUAS, 2015-07-27:
Using Paparazzi version v5.5.1_testing-0-g8756d2e-dirty with regular example/bebop.xml GCS height goes up and keeps counting up. 

Looking at baro messagesit all  all look fine. Noticed however lots of flips between 3D fix2D fix and None(also  outside

OpenUAS, 2015-07-27:
Blocked for Bebop not 5.6. A bebop is not really flyable without a good baro. Under investigation as of 20150727 - 13:30 CET Trying to pinpoint cause now.

OpenUAS, 2015-07-27:
Some findings The v-speed m/s speed also counts up to unlimited
GCS have 3D fix but no GPS (N/A)accuracy note impleneted
MESSAGES

GPS_INT:
eceef_z bit high but stable (5002200 m)

alt - stable
hsml - stable

pacc - 0 m
sacc - 0 m/s
fix - 3D (stable)

INS:
ins_z goes up and up
ins_zd goes up and up

Maybe I find some time today to look in the source...

OpenUAS, 2015-07-27:
Added
...

  <section name="INS" prefix="INS_">

...

     <define name="USE_GPS_ALT" value="TRUE"/>
    <define name="VFF_R_GPS" value="0.01"/>
  </section>

...

OpenUAS, 2015-07-27:
That "kindoff works" but I'm worried about the "Target altitude" going up before this "fix".. Will look at QNH and what not now..

fvantienen, 2015-07-27:
Fixed with cc51fcfad7022112bf2c4406101c96888d215165

closed 2015-07-27 by fvantienen