paparazzi/paparazzi #1266: [INS] USE_INS_NAV_INIT causes bad initialization of INS

EwoudSmeur, 2015-07-27:
USE_INS_NAV_INIT is default true, but gives problems when the flight plan lat0/lon0 are not close to the drone startup location.

I took a standard flight plan, which had it's origin at ENAC. I didn't take the effort to change the origin because with all relative waypoints they will move to my drone's position when I have a GPS fix anyway. However, the drone initially estimates an altitude of ~8000m, and only after the geo-init block comes back to zero. At this point, the vf_extended_float can have an accelero bias of up to 2m/s^2, which takes a very long time to go down and has it's effect on the estimated vertical speed.

If I set USE_INS_NAV_INIT to false I don't have problems, even though the lat0/lon0 in the flight plan are still wrong. The accelero bias is close to zero.

Until someone fixes this, we might consider putting USE_INS_NAV_INIT to false for default.

flixr, 2015-07-27:
What exact paparazzi_version are you using?
I fixed something like that some time ago: https://github.com/paparazzi/paparazzi/commit/b7cb5fd39aecc7989003ccabdeafe059e3aa16fa
Are you sure that it is not already fixed in master/v5.5.x_testing?

EwoudSmeur, 2015-07-27:
I merged with master on 18th of june, so I should have this commit. I forgot to check if this problem still exists in newest master, but I can do this later today.

OpenUAS, 2015-07-27:
I experience something like this in v5.5.1_testing-0-g8756d2e-dirty If I'm not mistaken.

flixr, 2015-07-27:
Yes, plz retest in latest v5.5.1_testing-5-g84a9fcb.
I can't reproduce the issue... After `ins_reset_local_origin` (called after geo_init) the bias seems to be correctly reset here.
if you can still reproduce it, please let me know which airframe (if it is in git), otherwise with flags you have set (using sonar, gps, etc.. ?)

EwoudSmeur, 2015-07-28:
Indeed, I don't see the problem now in master... though it is strange that in the GCS I see (before geo-init) an altitude of 8000 meter, while in rotorcraft_fp it is near zero

flixr, 2015-07-30:
So, can we close this or do you still see some issues?

closed 2015-08-24 by EwoudSmeur