paparazzi/paparazzi #1281: Since global waypoints update drones fly aat 45 meters

EwoudSmeur, 2015-08-03:
There has been a long lasting issue with drones flying in the roof of our optitrack arena. I looked into it today and what happens is that waypoints that are defined with lat/lon have an altitude offset of 45 meters. 

All waypoints have similar values in WAYPOINTS_ENU and WAYPOINTS_LLA, but the difference is their respective WAYPOINTS_GLOBAL flag. Waypoints that are defined relative with WAYPOINTS_ENU have this flag set to false and behave as expected. Waypoint that are set with lat/lon have this flag set to true and have their waypoints offset by 45m. 

I think this is quite unintuitive... How am I supposed to set a waypoint with lat/lon and still be able to specify the height relative to the initial point, like I used to? I looked at the code, but I can't make sense of the OCAML stuff...

flixr, 2015-08-03:
I don't quite understand...
The whole point of using global waypoints is to - well - have global coordinates for those. So the lat/lon/alt is global and will not change if with your local reference.

Could you please describe this in more detail, like how did you define the waypoints (do they use the default flight plan alt, have alt explicitly set or use height?).

EwoudSmeur, 2015-08-03:
If I set alt, what i get is alt +45. If I do nothing, I get alt +45 (irrespective of ground_alt).

I am sure that is the point of global coordinates, but I don't know who expects this... I am pretty sure **none** of the students in Delft expected this, as we have been getting drones out of the flight arena roof on a regular basis.

EwoudSmeur, 2015-08-03:
What I mean is: I am not expecting to get global coordinates. When I put in an altitude, how am I supposed to know that 45 meters will be added?

flixr, 2015-08-03:
I guess you are seeing the difference between altitude above MSL and above the WGS84 ellipsoid.

What you set in the flight plan is altitude above MSL...

flixr, 2015-08-03:
This seems to be a problem with optitrack/natnet that sends the same values for GPS alt (above WGS84 ellipsoid) and hmsl (height above mean sea level).

flixr, 2015-08-03:
Can we close this with the merge of #1284 then?

closed 2015-08-03 by EwoudSmeur