paparazzi/paparazzi #1288: [GCS] not updating map/ground speed when fix is not 3D

masierra, 2015-08-04:
Trying to have the gps system recognize higher levels of gps fix and assigning them higher values than 3. I am not exactly sure what is stopping the GCS from updating the map or ground speed when gps mode != 3D.

flixr, 2015-08-04:
The update of the aircraft position on the map is independent of the GPS fix.
But depending on which INS filter the aircraft uses it will not update its own position without GPS fix.

In the airborne code we require a 3D GPS fix, because experience showed that when you only have a 2D fix, the position is too inaccurate even when still on the ground.

OpenUAS, 2015-08-05:
@masierra Your focus improving and looking at details is great. I would advise to read into the matter and try to describe the exact proposed benefit and even better add a **detailed** description if what ou would lke to see improved nad if possible even start improving and if you tested it and gives better flight or flight experience post a link to your improved branch so we can start testing in live scenarios. TIA

masierra, 2015-08-06:
I'm currently working at Transition Robotics to implement Ardupilot-esque support for running multiple gps units simultaneously. We've been looking at Piksi RTK units which report not only a 3D fix but also two different RTK fixes as well. 

We found that while Piksis perform great once they have RTK, they take a bit to establish fix and they can loose fix as well.  For this, I am trying to use a ublox as a backup and so support for two modules at once is underway.

OpenUAS, 2015-08-18:
Thx for explaining in detail, now it is clear and also the benefit is obvious :+1:

flixr, 2015-10-14:
@masierra Since #1349 was merged we have additional DGPS and RTK fix modes for GPS, so is this still an issue or can we close this?

flixr, 2015-10-28:
@masierra so this can be closed then after your fixes from https://github.com/paparazzi/paparazzi/pull/1386 ?

closed 2015-10-29 by masierra