paparazzi/paparazzi #1295: 3D gps fix

devyeshtandon, 2015-08-09:
The Flight Recorder GPS does not work even after editing (airframes/examples/ardrone2.xml) the protocol to sirf and removing the gps_ubx_ucenter module from the modules loading block.

flixr, 2015-08-25:
@OpenUAS @fvantienen could you guys plz test this?

OpenUAS, 2015-08-25:
Yes, will do.

OpenUAS, 2015-08-25:
---[ end trace b38fe2281818d79d ]---
Enabling CTS wakeup for UART1
scsi 0:0:0:0: Direct-Access     Generic  Ultra HS-SD/MMC  1.76 PQ: 0 ANSI: 0
usb-storage: device scan complete
sd 0:0:0:0: [sda] 7733248 512-byte logical blocks: (3.95 GB/3.68 GiB)
sd 0:0:0:0: [sda] Write Protect is off
sd 0:0:0:0: [sda] Mode Sense: 23 00 00 00
sd 0:0:0:0: [sda] Assuming drive cache: write through
sd 0:0:0:0: [sda] Assuming drive cache: write through
 sda:
sd 0:0:0:0: [sda] Assuming drive cache: write through
sd 0:0:0:0: [sda] Attached SCSI removable disk
usb 1-1.1:1.0: uevent
usb 1-1.1: uevent
usb 1-1.1:1.0: uevent
usb 1-1.1:1.0: uevent
channel hint set to 0
Add Filter 0 = 01:00:5e:00:00:01
Keep Filter 0 = 01:00:5e:00:00:01
ov7670 2-0021: Enabling Streaming
Parrot Micronesie GPS : with anti-Nack
Enabling CTS wakeup for UART2
DSPTrace:H264Enc Create
DSPTrace:H264Enc Create Done
------------[ cut here ]------------
WARNING: at /home/stephane/.ardrone/linux/ardrone2_ARDrone2_Version_20130102/Linux/kernel/omap/kernel/mutex.c:207 __mutex_lock_slowpath+0x70/0x294()
Modules linked in: ov7670 soc1040 omap3_isp ar6000
[<c0039638>] (unwind_backtrace+0x0/0xcc) from [<c0069510>] (warn_slowpath_common+0x48/0x60)
[<c0069510>] (warn_slowpath_common+0x48/0x60) from [<c035f56c>] (__mutex_lock_slowpath+0x70/0x294)
[<c035f56c>] (__mutex_lock_slowpath+0x70/0x294) from [<c035f79c>] (mutex_lock+0xc/0x20)
[<c035f79c>] (mutex_lock+0xc/0x20) from [<c01d1b5c>] (tty_throttle+0x14/0x58)
[<c01d1b5c>] (tty_throttle+0x14/0x58) from [<c01d0ef0>] (n_tty_receive_buf+0xe8c/0xec0)
[<c01d0ef0>] (n_tty_receive_buf+0xe8c/0xec0) from [<c01d3650>] (flush_to_ldisc+0x10c/0x1c0)
[<c01d3650>] (flush_to_ldisc+0x10c/0x1c0) from [<c01e5cfc>] (serial_omap_irq+0x568/0x6dc)
[<c01e5cfc>] (serial_omap_irq+0x568/0x6dc) from [<c0093270>] (handle_IRQ_event+0x34/0xf0)
[<c0093270>] (handle_IRQ_event+0x34/0xf0) from [<c0094b7c>] (handle_level_irq+0xc0/0x150)
[<c0094b7c>] (handle_level_irq+0xc0/0x150) from [<c0033068>] (asm_do_IRQ+0x68/0x84)
[<c0033068>] (asm_do_IRQ+0x68/0x84) from [<c0033a84>] (__irq_svc+0x44/0xa8)
Exception stack(0xc0493f40 to 0xc0493f88)
3f40: 003a3529 00000000 003a3529 00000000 c04d3694 c04d365c 00000003 00000003
3f60: c0529a48 413fc082 00000000 00000000 00000000 c0493f88 c0048718 c0048724
3f80: 60000053 ffffffff
[<c0033a84>] (__irq_svc+0x44/0xa8) from [<c0048724>] (omap3_enter_idle+0x11c/0x150)
[<c0048724>] (omap3_enter_idle+0x11c/0x150) from [<c028bde4>] (cpuidle_idle_call+0xac/0x104)
[<c028bde4>] (cpuidle_idle_call+0xac/0x104) from [<c00351c0>] (cpu_idle+0x34/0x8c)
[<c00351c0>] (cpu_idle+0x34/0x8c) from [<c000887c>] (start_kernel+0x244/0x298)
[<c000887c>] (start_kernel+0x244/0x298) from [<80008034>] (0x80008034)
---[ end trace b38fe2281818d79e ]---
Enabling CTS wakeup for UART2
Enabling CTS wakeup for UART2
Parrot Micronesie GPS : with anti-Nack
ecc correction in bits 4113,
Parrot Micronesie GPS : with anti-Nack
ecc correction in bits 833,
ov7670 2-0021: Disabling Streaming
Enabling CTS wakeup for UART2
Spurious irq 95: 0xffffffdf, please flush posted write for irq 56

flixr, 2015-08-25:
@OpenUAS wut?? ublox gps is not working in v5.6 but in v5.7? I can hardly believe that since there are no relevant changes...
Can you plz provide the exact version where it is not working?

devyeshtandon, 2015-08-25:
@OpenUAS, using paparazzi ver 5.6.. 
Can anyone enlighten me what is ATM and how do we generate it? Also, how do we find where the exact flaw is?

flixr, 2015-08-25:
@devyeshtandon can you plz provide the exact version you are using?
https://wiki.paparazziuav.org/wiki/FAQ#How_do_I_check_which_Paparazzi_Version_I.27m_using

devyeshtandon, 2015-08-25:
v5.5.1_testing-0-g8756d2e-dirty

OpenUAS, 2015-08-25:
@devyeshtandon ATM = AT This Moment.  :)

OpenUAS, 2015-08-25:
@flixr Will do a retest 5.6 tommorrow, will now focus on SIFT ,  sirf_parse_char is reached however c= 0xA0 never reached exept just before I press a second time CTRL-C a quick n dirty Printf gave 

...

^Cc == 0xA0 - c == 0xA0 - c == 0xA0 - Received SIGINT
Press Ctrl-C again to stop the program
^CReceived SIGINT
Leaving now

...

flixr, 2015-08-25:
@devyeshtandon thx... you should update to the v5.6 stable branch in any case...

OpenUAS, 2015-10-14:
Can retest, not that I like the dongle so much but before ARDrone2 is obsolete like to have essential fixed one and for all. Then after frontcam support for bebob, ARdrone2 is likely not used nor changed all to much anymore...

flixr, 2015-11-03:
Shall we drop support for the flight recorder?
Unless actually has an ARDrone2 with FlightRecorder and is willing to fix this, we should probably close this as WONT_FIX and removing the code for this (as it currently doesn't work).

devyeshtandon, 2015-11-04:
I have both of them and I am willing to fix this issue, but I am not sure how to go about it. I am afraid, I have never worked on such kind of thing before.

dewagter, 2015-11-04:
We did not implement it because the GPS is not very good. We recommend
ublox.

flixr, 2015-11-04:
AFAIK it was working at some point... but if you can, definitely use a ublox instead...

OpenUAS, 2015-11-20:
Assigned to self since there still will be people in next years want to try PPRZ with this thing, and it is jst out of curiosity if it is at all possible to fly from A to B with it.  I will more be like a Christmas time type of thing to work on a bit. So no deadline yet.. but plz leave issue open till spring.

OpenUAS, 2016-03-22:
Well more like a eastern time type of thing to work on then :( ... Since uBlox modules with USB can already be had for very low price by now... time to close it.

@devyeshtandon feel free to reopen the issue and use it as a tryout to dive into PPRZ for yourself

closed 2016-03-22 by OpenUAS

pache1211, 2016-04-05:
Does the sirf mode working for the parrot flight recorder??

flixr, 2016-04-07:
@OpenUAS was this actually ever working?
In any case if you guys don't plan on putting effort into this we should mark it as WONTFIX as nobody else from the core Paparazzi devs has this stuff anyway...
Although pull requests actually fixing it would of course be welcome from anyone.

pache1211, 2016-04-08:
I manage to get GPS fix. I haven't done a flight test, but to get the GPS fix I have to download sirf demo and configure the flight recorder to sirf mode, since it seems that the flight recorder is working in NMEA at 4800. So configure it to SIRF mode to 57600, and make the changes in the airframes/examples/ardrone2.xml and I got the GPS fix

streamholder, 2016-08-13:
@pache1211 how did you manage to change the configuration of the Flight Recorder? I'm not able to connect to it from SiRFDemo using Parrot's driver on Windows 10.
If you can, contact me in private at flightrecordercontact [at] inboxalias.com (sorry about the alias, I'll give you my real address as soon as you contact me) as I'm working on a adding support for the Flight Recorder in Paparazzi as I don't think the receiver works as badly as they claim on the Wiki.