paparazzi/paparazzi #1320: Multi-Aircraft Support for Binding Keys/Papget Buttons

hooperfly, 2015-08-28:
# Current Capabiltiy

Currently any individual *flight plan* can bind a keyboard `key` to a given flight block action. For example:

```
  <block key="D" name="descent" strip_button="Descent">
    <circle wp="HOME" throttle="0.0" pitch="-15" vmode="throttle"/>
  </block>
```

Reference wiki:   http://wiki.paparazziuav.org/wiki/Flight_Plans#Blocks

This works for only a single aircraft and is limited to *flight plan* `blocks`. 

# Proposed Extensions/Enhancements

It would be nice to extend the `papget` feature to allow `key` bindings by extending the `papget` attributes. The addition of the `key` attribute/value pair is a possible mechanism for binding keys to multi-aircraft events. Currently, `papgets` can apply to a single aircraft or across all aircraft in a multi-aircraft environment. 

```
  <property name="key" value="t"/>
```

Reference wiki: http://wiki.paparazziuav.org/wiki/GCS#Buttons

NOTE: It might be nice to extend the papget block to accepts a list of AC_IDs to facilitate sending messages to a subset of aircraft in a group but that is another `issue` to resolve.

```
  <property name="ac_id" value="10, 15, 9"/>
```

Reference wiki: http://wiki.paparazziuav.org/wiki/GCS#Aircraft_binding

gautierhattenberger, 2016-10-12:
Is this still relevant ? Papgets buttons should now be compatible with multi-aircraft. Not sure add key binding feature is really a priority.

hooperfly, 2016-10-13:
Can you bind `keys` to the papgets? Also, papgets apply to either 1 or all aircraft. As mentioned, in the `Related Note` above, it would be nice to have the ability to bind papgets to a subset of aircraft, 1 < ac_id_list < all_ac_id

gautierhattenberger, 2018-06-06:
closing as won't be fixed in current GCS
see https://github.com/paparazzi/paparazzi/projects/2

closed 2018-06-06 by gautierhattenberger