paparazzi/paparazzi #1331: Paparazzi Center: On BUILD the list of settings should be updated

dewagter, 2015-09-04:
When a module is changed in the airframe file, one needs to manually change to another airframe and back in the list of airframes, or restart paparazzi center in order for the correct settings to be listed.

When changing/removing a module which has settings in an airframe file, and then clicking build/upload, it compiles the module-settings of the modules that were previously in the list and the build fails.

flixr, 2015-09-21:
@gautierhattenberger any idea on how to reload the config by calling update_params (part of the ac_combo_handler closure) from the build_handler?

gautierhattenberger, 2015-09-22:
see `reload_on_build` branch

dewagter, 2015-09-22:
Wow, that's a lot of code ;-)

gautierhattenberger, 2015-09-22:
yeah, I think we can add the upload button without too much work if you really want :p

flixr, 2015-09-22:
Lol, I clearly didn't have enough sleep when I added the refresh button and tried to connect it to the build button as well (and somehow failed)...

closed with 2843a415d6a9bf1a87134e5445f7f3686f174876

flixr, 2015-09-22:
Ahrg, ok... maybe I did have enough sleep after all...

flixr, 2015-09-22:
The problem doing it this way is that the build starts before the config is reloaded... that is why I thought it needs to be integrated in the build_handler or somewhere like that...

gautierhattenberger, 2015-09-22:
do you have an idea of the order of the callbacks in GTK ?

flixr, 2015-09-22:
no idea, but it doesn't work like this...
And IMHO the only proper solution to guarantee this is to add the refresh in the build button callback...

gautierhattenberger, 2015-09-22:
I'll try, but from what I see, it should work, they are called in register order

flixr, 2015-09-22:
Actually it seems that even using the reload button, it doesn't always work:
Steps to reproduce:
- e.g. take the bebop.xml and remove the air_data module
- click reload and then build

While it reloads the conf and displays it correctly in paparazzi center, this doesn't update the settings in the conf which is used to generated the settings header, that still only happens when changing the airframe file...

gautierhattenberger, 2015-09-22:
updated in `reload_on_build` branch.
it updates the display and save the conf as well.
the callbacks order is correct.

flixr, 2015-09-22:
Thx, seems to work now, pushed to master and also added the reload on upload.

Seems we can now really close this with bf5c3ce58bb183cb8c351e326f59f274648890c1 and 5659d82c887c8c7c05de8c209167ebb58cedc46f

closed 2015-09-22 by flixr