paparazzi/paparazzi #1344: server: doesn't send DL_VALUES if there are a lot of settings

flixr, 2015-09-12:
When you add more settings, at some point server doesn't send DL_VALUES anymore with
`Discarding long ivy message DL_VALUES (1240 bytes)`

What is this Ivy bug the limit of 1000 chars in `message_send` in pprz.ml is supposed to prevent?

flixr, 2015-09-21:
@gautierhattenberger @alex31 any idea why there is this limitation?

gautierhattenberger, 2015-09-21:
Actually, Ivy should support long messages since 2008, but Pascal added this check in 2009. We should try to remove it and see because their is no reasons to keep it from what I can see.

flixr, 2015-09-22:
So what is the maximum length now?
I guess it would still make sense to prevent endlessly long messages?

gautierhattenberger, 2015-09-22:
there is an initial malloc and a realloc if needed, so I guess the limit is the available memory... but yes, limiting stupidly long message is probably wise
the initial alloc is around 4000 bytes I think.

flixr, 2015-09-22:
So we limit it to 4000 or 10000 or so?

gautierhattenberger, 2015-09-25:
tested and working

PR #1364 (2015-09-25): [ocaml] increase maximum Ivy message length from 1000 to 10000
should fix #1344
commit 2635a908 (2015-09-25): [ocaml] increase maximum Ivy message length from 1000 to 10000

should fix #1344
closed 2015-09-25 by gautierhattenberger