paparazzi/paparazzi #1368: Absolute waypoints get lost by aircraft build (both ap & nps)

tcunis, 2015-09-28:
Using v5.7_devel build, building an airframe for target ap or nps dismisses waypoints defined absolutely in flight-plan xml:
For instance,
```xml
<waypoints>
    <waypoint lat="51.990630" lon="4.376823" name="p1"/>
</waypoints>
```
(AC_flightplan.xml) becomes
```C++
#define WAYPOINTS { \
 {0.0, 0.0, 1},\
};
#define WAYPOINTS_ENU { \
 {0.00, 0.00, 1.00}, /* ENU in meters  */ \
};
```
and so forth (*aircrafts/AC/(ap|nps)/generated/flight_plan.h*).

flixr, 2015-09-29:
Hi @tcunis 

I don't think there any issue here. I would bet that your waypoint p1 has the same coordinates as your flight plan (lat0, lon0).
`WAYPOINTS` and `WAYPOINTS_ENU` contain the local coordinates (in UTM and ENUrelative to your flight plan location.
`WAYPOINTS_LLA` contain the same waypoints in absolute (LLA) coordinates.

If you still found a problem, could you please provide more details?
Llike the exact version (output of ./paparazzi_version), complete flight plan xml file and compile output?

tcunis, 2015-09-29:
Hey @flixr ,
no, it doesn't. There are couple of waypoints encircling the flight-plan center which are generated to (0,0).

flixr, 2015-09-29:
Ok, can you then plz provide more details so we can reproduce it?

flixr, 2015-09-29:
Thx for reporting, I think I found it :-)

closed 2015-09-29 by flixr

tcunis, 2015-09-29:
Yet I've just got the detail files out of the virtual machine... For some reason, I cannot upload any files to github.com.

If only for completeness:
> cunis@vtc-1:~/paparazzi$ ./paparazzi_version 
v5.7_devel-326-ga776aac

commit aed91422 (2015-09-29): [generators] fix generation of local waypoints

error was accidentally introduced in 44111be59ofda0ed46961205d04c84fd3ec58c9fe

closes #1368