paparazzi/paparazzi #1395: GCS - Papgets not visible on GCS screen : Mac OS X El Capitan

hooperfly, 2015-10-29:
It seems that the homebrew gtk library now runs natively on Mac OS X as of El Capitan. As a result, `papgets` are not visible on the GCS but they are active. The place/location where a given papget is placed will activate the appropriate action when pressed but the `papaget` icon is not visible.

Screen capture of GCS using GTK+ Quartz library under El Capitan( papgets not showing )

[image]

Screen capture of GCS using GTK+ X11 library under Yosimite

[image]

hooperfly, 2015-10-29:
It looks like there were quite a few changes to papgets related code on Sept. 27th. This could be related to those changes as well.

https://github.com/paparazzi/paparazzi/search?utf8=%E2%9C%93&q=papget

gautierhattenberger, 2018-06-06:
closing as won't be fixed in current GCS
see https://github.com/paparazzi/paparazzi/projects/2

closed 2018-06-06 by gautierhattenberger