paparazzi/paparazzi #1396: GCS - Google Maps not loading - Mac OS X El Capitan

hooperfly, 2015-10-29:
Google maps views are not being displayed in the GCS using the GTK+ Quartz library in tandem with Mac OS X El Capitan.

hooperfly, 2015-11-01:
After many hours of poking around various mail threads related to gtk+, cairo, ocaml, libgnomecanvas and issues related to Mac OS X El Capitan, it seems this issue might be related to bugs in El Capitan. There were many "under the hood" changes to "improve" performance with El Capitan and it seems it's fairly buggy. Stay tuned....

OpenUAS, 2015-11-01:
Hats off for your investigation, appriciated!

hooperfly, 2015-11-02:
Decided to become a Beta tester for Mac OS X so we can get early releases of updates as well as the ability to give feedback to Apple about the issues. Downloading 10.11.2 Beta as I type this message. More to come.....

hooperfly, 2015-11-11:
Submitted a bug report to the OCaml Forge site related to `lablgtk`, Mac OS X, GTK/Quartz issues:

https://forge.ocamlcore.org/tracker/index.php?func=detail&aid=1487&group_id=1&atid=101

OpenUAS, 2015-11-20:
Thx, the other no screen redraw issue you've posted likely related

OpenUAS, 2016-12-21:
@hooperfly any news on this?

wv-tud, 2017-04-04:
I can report this is still not working under OSX 10.12.4 (macOS Sierra)

gautierhattenberger, 2018-06-06:
closing as won't be fixed in current GCS
see https://github.com/paparazzi/paparazzi/projects/2

closed 2018-06-06 by gautierhattenberger