paparazzi/paparazzi #1413: Vote for feature

dewagter, 2015-11-11:
Test

flixr, 2015-11-11:
test  this as well :+1:

rmeertens, 2015-11-11:
Did you consider the fact that when having more than two options there is no good voting system? Please consider Arrow's impossibility theorem:  https://en.wikipedia.org/wiki/Arrow's_impossibility_theorem

flixr, 2015-11-11:
We don't have to use it if there are more than two options....

rmeertens, 2015-11-11:
So in this issue we can choose for:
- voting for everything
- never voting
- only voting if there are only two options. 

mmm,

dewagter, 2015-11-11:
If you know a system to vote for 3 or more would be nice:

a) absolutely a good idea (changing signs on spektrum transmitters)
b) no strong opinion (if it makes some people happy to have the correct sign on their transmitter display I'm ok to reverse 3 channels)
c) terribly against (I just sold hundreds of ladybirds and do not want my customers to need to change transmitter signs even if a single define USE_OLD_SPEKTRUM_REVERSED_SIGNS in the airframe can solve the problem)

closed 2015-11-12 by dewagter