paparazzi/paparazzi #1428: Fresh Install compile error

coppolam, 2015-11-18:
Command line output when using "make" after "make clean".

------------------------------------------------------------
Building Paparazzi version v5.7_devel-597-g2258100
------------------------------------------------------------
make -C data/maps
mkdir /home/mario/paparazzi/nexport/conf/maps_data
mkdir: cannot create directory ‘/home/mario/paparazzi/nexport/conf/maps_data’: No such file or directory
make[1]: *** [/home/mario/paparazzi/nexport/conf/maps_data] Error 1
make: [update_google_version] Error 2 (ignored)
[ -L conf/conf.xml ] || [ -f conf/conf.xml ] || cp conf/conf_example.xml conf/conf.xml
[ -L conf/control_panel.xml ] || [ -f conf/control_panel.xml ] || cp conf/control_panel_example.xml conf/control_panel.xml
make -C sw/lib/ocaml
OCY expr_parser.mly
OCL expr_lexer.mll
40 states, 423 transitions, table size 1932 bytes
DEPEND .depend
OC fig.mli
OC fig.ml
OC debug.mli
OC debug.ml
OC base64.mli
OC base64.ml
OC serial.mli
OC serial.ml
OC ocaml_tools.mli
OC ocaml_tools.ml
OC expr_syntax.mli
OC expr_syntax.ml
OC expr_parser.mli
OC expr_parser.ml
OC expr_lexer.ml
OC extXml.mli
OC extXml.ml
OC env.mli
OC env.ml
OC xml2h.mli
OC xml2h.ml
OC latlong.mli
OC latlong.ml
OC egm96.mli
OC egm96.ml
OC srtm.mli
OC srtm.ml
OC http.mli
OC http.ml
OC maps_support.mli
OC maps_support.ml
OC gm.mli
OC gm.ml
OC iGN.mli
OC iGN.ml
OC geometry_2d.mli
OC geometry_2d.ml
OC cserial.c
OC convert.c
OC ubx.mli
OC ubx.ml
OC pprz.mli
OC pprz.ml
OC xbee.mli
OC xbee.ml
OC logpprz.mli
OC logpprz.ml
OC xmlCom.mli
OC xmlCom.ml
OC os_calls.mli
OC os_calls.ml
OC editAirframe.mli
OC editAirframe.ml
[... 419 lines cut ...]
OC gps.mli
OC gps.ml
OC sim.mli
OC sitl.mli
OC sitl.ml
OC sim.ml
OL sitl.cma
make -C sw/ground_segment/joystick
DEPEND .depend
CC test_sdl_stick.o
BUILD test_stick
CC sdl_stick.o
CC ml_sdl_stick.o
BUILD sdl_stick.so
OC input2ivy.ml
OL input2ivy
GENERATE /home/mario/paparazzi/nexport/var/include/messages.h
GENERATE /home/mario/paparazzi/nexport/var/include/ubx_protocol.h
GENERATE /home/mario/paparazzi/nexport/var/include/mtk_protocol.h
GENERATE /home/mario/paparazzi/nexport/var/include/xsens_protocol.h
GENERATE /home/mario/paparazzi/nexport/var/include/dl_protocol.h
GENERATE /home/mario/paparazzi/nexport/var/include/abi_messages.h
GENERATE /home/mario/paparazzi/nexport/var/include/mavlink/
make: *** /home/mario/paparazzi/nexport/sw/ext: No such file or directory.  Stop.
make: *** [/home/mario/paparazzi/nexport/var/include/mavlink/protocol.h] Error 2

flixr, 2015-11-18:
Seems like you have exported `PAPARAZZI_SRC` and/or `PAPARAZZI_HOME` incorrectly.
Either remove these env variables or set them correctly.

OpenUAS, 2015-11-19:
May bad, open a terminal window then

$ nano ~/.bashrc

Look for PAPARAZZI_SRC and PAPARAZZI_HOME and correct the directories found there.

When correct will look like 

 export PAPARAZZI_HOME=~/paparazzi
 export PAPARAZZI_SRC=~/paparazz

closed 2015-11-19 by flixr