paparazzi/paparazzi #1430: GCS fails to show the correct location if a wrong flight plan is selected

EwoudSmeur, 2015-11-20:
When I upload the dummy flight plan when I am in Delft, the GCS shows a wrong location in the GCS, both in the maps section as well as the latitude and longitude in the top right part of the screen.

When I upload a flight plan with lat0 and lon0 in delft and a geo_init, I don't have this problem.

In both cases I have a GPS 3D fix and telemetry set to default. The GPS location is sent to the ground station, so it seems odd that the GCS would give a different (wrong) position.

Does this have to do with the position being calculated relative to a ref location that is set based on the lat/lon in the flight plan and reset upon geo_init?

flixr, 2015-11-20:
This is expected.... unless you do a geo init, the initial location well be the lat0/lon0 specified in the flight plan (if `USE_INS_NAV_INIT` is TRUE, which is the default).

Since the dummy flight plan doesn't have a geo_init, it will never set another origin...

Or am I misunderstanding your problem?

EwoudSmeur, 2015-11-20:
Shouldn't the gcs show the correct position regardless of what you fill in in the flight plan if there is gps available? I mean... the value is there, why take something out of the flight plan?

gautierhattenberger, 2015-11-20:
The mapCanvas object that holds the 2D map of the GPS needs to be initialized with a reference position at some point. I have no idea how difficult it would be to change that, and if it is really a problem after all. A few elements (handled by the server) also need a reference since they are sent with local coordinates by the aircraft.
The source of the issue is most likely the fact that the airborne code needs to use local coordinates at some point. For instance on rotorcraft you need to at least initialized the ref point to compute LTP frame, and it has to be close to the place you are flying. For several reasons, the default is to use the flight plan default and don't init automatically to your current position.
You can either set `USE_INS_NAV_INIT` to false or do a geo init in your flight plan.

flixr, 2015-11-21:
Would like to add:
The GCS doesn't show the exactly GPS position, the GPS messages is mainly sent for information/debug purposes (and the timestamp).
The message that is used to display position is [ROTORCRAFT_FP](http://docs.paparazziuav.org/latest/paparazzi_messages.html#ROTORCRAFT_FP), which contains the position from the state interface.
The position in the state interface is set by the chosen INS algorithm, so unless you use the "gps_passthrough" type this is not directly your GPS position.

EwoudSmeur, 2015-11-25:
Thanks @gautierhattenberger and @flixr  for your replies. I agree that this is probably not the most pressing of issues.

But I would still say there is something strange going on. The ins (local) coordinates are calculated using the gps (and perhaps other sensors), using some calculation. When these coordinates are converted back to global coordinates in the gcs the location suddenly does not correspond with gps anymore. I would guess that a different calculation is used to go from global gps -> local ins than for the reverse calculation.

flixr, 2015-11-27:
@EwoudSmeur I think the "only issue" here is that you really can't use a local tangent plane reference far away from where you actually are, as the assumption made there is that the earth is flat... which it is obviously not.
So if you set your LtpDef (origin of local cartesian coordinate system) to some specific location (which at startup is the flightplan location if `USE_INS_NAV_INIT` is not explicitly set to FALSE) and then get a GPS location from somewhere totally else (where you really are), this will convert your current location (given in ECEF coordinates) to a local position w.r.t. the LtpOrigin under the flat earth assumption.
Of course this won't result in correct coordinates...

Hope that makes it more clear?

Not sure we can really do something about this in such a case... at least I can't think of something.
The filter estimates are in a cartesian LTP frame, and if the origin of that frame is far away... well.. don't do that.
In your case it should be enough to "simply" set `USE_INS_NAV_INIT` to false so it will take the first GPS position it gets to set the local origin...

flixr, 2015-11-27:
So to be clear: this has nothing to do with the GCS directly, rather it is because the INS you are using is estimation position in a local cartesian coordinate frame (which by default is initialized at flight plan location).

flixr, 2015-11-27:
One last comment that hopefully finally clears up where the problems are with having the origin of the cartesian LTP frame far away:
Maybe what you were seeing is an overflow of the local position: since it is represented using a int32 with `INT32_POS_FRAC`, the max range of that var is ~8388km per axis.
So if your LTP origin is further away it will definitely overflow/wrap, giving you a completely wrong location...
But even before that happens, you will run into problem since the "flat earth" assumption really doesn't hold anymore over larger distances...

So to sum up: if your "dummy" flight plan has a location with maximum a couple of hundred kilometers away, the horizontal pos should be roughly correct, but the altitude probably be way off.
If you are really far away from your origin frame, the representation will even overflow and show you a completely wrong location...

In any case: if you use an INS that estimates position in a local frame, set the origin of that local frame very close to where you actually are.
Either by using geo_init, or if you don't use that set a close location in your flight plan or set `USE_INS_NAV_INIT` to FALSE.

closed 2015-11-27 by flixr

dewagter, 2015-11-27:
I'm still wondering exactly which "LTP assumption/simplifications" are made
in the code.

I used a lot of LLH / ECEF / LTP combinations in the past and could convert
from any one to any other without any "error".

I know that if your LTP origin is e.g. at the equator, then in Delft you
have weird coordinates (e.g. half the earth radius negative altitude, with
xy not parallel to ground). I'm not saying it's meaningful but given the
origin you can still convert it to correct LLH, so if it's not something
must be over-simplified somewhere.

Overflow obviously is a different matter.

-Christophe

flixr, 2015-11-27:
I'm not aware of any "over-simplifications" in the coordinate transformations per se.
As far as I can see (and could confirm in some quick basic tests), ff you "simply" convert the ECEF pos from the GPS to NED using the LtpDef and send that to the GCS, it will display the position correctly.
Of course only if you are not too far away (max couple of hundereds of km), so that you don't get any overflows in the representation (or the intermediate computed values).

The implementation of the ins_int is a bit of another matter (but that has nothing to do with the GCS)...
There the assumption of our local frame is that for the whole area you fly in, that z points down, x to north (It is not a LTP frame that moves with the aircraft) and the earth curvature doesn't really play a role in altitude...
Especially if you use the baro alt and not the GPS alt (ECEF converted/rotated to NED) the altitude will not be correct if you are far away from the origin.

I couldn't find anything wrong, but please make your own tests to check....
But using the dummy flight plan in the repo (location Toulouse) without geo init or `USE_INS_NAV_INIT=FASE` in Delft will result in wrong positions due to the limited range of the representation. Choosing something closer (e.g. lat0="50.0" lon0="4.0") in the flight plan was showing correct (horizontal) position here... but that is still further away than what you should use as the flat earth assumption in the filter definitely can't be made anymore.

gautierhattenberger, 2015-11-28:
@dewagter @EwoudSmeur can you try to add GPS_LLA message to the periodic telemetry and comment the lines that update the position in `rotorcraft_server.ml` file (ROTORCRAFT_FP message). It's not really a solution, I'm just wondering if it solves your issue.