paparazzi/paparazzi #1432: Rotorcraft FBW cannot fly it in sim

OpenUAS, 2015-11-20:
As a user I'll like to fly extra safe and use FBW, but I cannot sim the flight fully and I do want to use NPS.

dewagter, 2015-11-21:
You can simulate a quadcopter (rotorcraft_mixin) but not a helicopter
(pwm_direct).

But this is not related to FBW, unless your point is to simulate the
intermcu subsystem.

-Christophe

OpenUAS, 2015-11-23:
Was this not then the open issue that needed to be fixed if intermcu pull request was honoured?, If issue to fix was misunderstood please close this issue.

flixr, 2016-02-27:
So if I understand this correctly, then this is about a missing helicopter model for e.g. the JSBSim FDM backend? Then I would suggest we close this issue and maybe open a new feature request for a helicopter model...
Or is there really some issue with simulating split ap/fbw multicopters in NPS?

dewagter, 2016-03-03:
I think it's mainly that airframes with fbw do not yet have a working
sim/NPS target. Might be possible to just add it, but might need some extra
code e.g. in case of helicopter-mixed actuators.

flixr, 2016-03-03:
So are there any airframes with fbw that we would have a JSBSim model for?
Then we could simply add the nps target there and close this issue...

dewagter, 2016-03-07:
not many, but these 2 for instance do not have a nps target:

https://github.com/paparazzi/paparazzi/blob/master/conf/airframes/TUDELFT/tudelft_logo600.xml
https://github.com/paparazzi/paparazzi/blob/master/conf/airframes/OPENUAS/openuas_vivify.xml

the new quadrotor FBW can already be simulated:

https://github.com/kevindehecker/paparazzi/blob/PX4/conf/airframes/TUDELFT/tudelft_iris_pid.xml

-Christophe

flixr, 2016-03-07:
well, logo600 is a conventional helicopter frame for which we don't have a JSBSim model...
For vivify we also don't have a nicely matching fixedwing model...

closed 2016-03-07 by flixr