paparazzi/paparazzi #1435: [python] messages/settings not working with wxgtk3.0

flixr, 2015-11-21:
From message app:
>
> Traceback (most recent call last):
>   File "/usr/lib/python2.7/dist-packages/wx-3.0-gtk2/wx/_core.py",
> line 16763, in <lambda>
>     lambda event: event.callable(*event.args, **event.kw) )
>   File
> "/home/garciafa/paparazzi/sw/ground_segment/python/messages_app/messagesframe.py",
> line 149, in gui_update
>     self.add_new_message(aircraft, msg.msg_class, msg.name)
>   File
> "/home/garciafa/paparazzi/sw/ground_segment/python/messages_app/messagesframe.py",
> line 137, in add_new_message
>     grid_sizer.AddGrowableCol(1)
>   File "/usr/lib/python2.7/dist-packages/wx-3.0-gtk2/wx/_core.py",
> line 15364, in AddGrowableCol
>     return _core_.FlexGridSizer_AddGrowableCol(*args, **kwargs)
> wx._core.PyAssertionError: C++ assertion "!IsColGrowable( idx )"
> failed at ../src/common/sizer.cpp(1976) in AddGrowableCol():
> AddGrowableCol() called for growable column

Which looks like API inconsistencies...

garciafa, 2015-11-23:
Here is the other error type for the messages app :
> Traceback (most recent call last):
>  File "/usr/lib/python2.7/dist-packages/wx-3.0-gtk2/wx/_core.py", line 16763, in <lambda>
>     lambda event: event.callable(*event.args, **event.kw) )
>   File "/home/garciafa/paparazzi/sw/ground_segment/python/messages_app/messagesframe.py", 
> line 82, in update_leds_real
>     aircraft.messages_book.SetPageImage(message.index, 0)
>   File "/usr/lib/python2.7/dist-packages/wx-3.0-gtk2/wx/_core.py", line 13560, in SetPageImage
>     return _core_.BookCtrlBase_SetPageImage(*args, **kwargs)
> wx._core.PyAssertionError: C++ assertion "page < GetPageCount()" failed at ../sr
> /gtk/notebook.cpp(276) in SetPageImage(): invalid notebook index
> Traceback (most recent call last):
>  File "/usr/lib/python2.7/dist-packages/wx-3.0-gtk2/wx/_core.py", line 16763, in <lambda>
>     lambda event: event.callable(*event.args, **event.kw) )
>  File "/home/garciafa/paparazzi/sw/ground_segment/python/messages_app/messagesframe.py", 
> line 155, in gui_update
>     aircraft.messages[msg.name].field_controls[index].SetLabel(msg.get_field(index))
> IndexError: list index out of range

For the setting app, here are the errors :
>Traceback (most recent call last):
>  File "/home/garciafa/paparazzi/sw/ground_segment/python/settings_app/settingsapp.py", line 58, 
> in <module>
>     main()
>   File "/home/garciafa/paparazzi/sw/ground_segment/python/settings_app/settingsapp.py", line 54, 
> in main
>     application = SettingsApp(0)
>   File "/usr/lib/python2.7/dist-packages/wx-3.0-gtk2/wx/_core.py", line 8628, in __init__
>     self._BootstrapApp()
>   File "/usr/lib/python2.7/dist-packages/wx-3.0-gtk2/wx/_core.py", line 8196, in _BootstrapApp
>     return _core_.PyApp__BootstrapApp(*args, **kwargs)
>   File "/home/garciafa/paparazzi/sw/ground_segment/python/settings_app/settingsapp.py", line 47, 
> in OnInit
>     self.main = settingsframe.create(None, options['ac_id'])
>   File "/home/garciafa/paparazzi/sw/ground_segment/python/settings_app/settingsframe.py", line 
> 16, in create
>     return SettingsFrame(parent, ac_ids)
>   File "/home/garciafa/paparazzi/sw/ground_segment/python/settings_app/settingsframe.py", line 
> 127, in __init__
>     slider = SettingCtrl(page, setting)
>   File "/home/garciafa/paparazzi/sw/ground_segment/python/settings_app/settingsframe.py", line 
> 65, in __init__
>     wx.Slider.__init__(self, parent=parent, minValue=min_v, maxValue=max_v, 
> style=wx.SL_HORIZONTAL | wx.SL_AUTOTICKS, size=(200, 30), id=setting.index + 
> SLIDER_ID_OFFSET)
>   File "/usr/lib/python2.7/dist-packages/wx-3.0-gtk2/wx/_controls.py", line 2827, in __init__
>     _controls_.Slider_swiginit(self,_controls_.new_Slider(*args, **kwargs))
> wx._core.PyAssertionError: C++ assertion "id == wxID_ANY || (id >= 0 && id < 32767) || (id >= 
> wxID_AUTO_LOWEST && id <= wxID_AUTO_HIGHEST)" failed at ../src/common/wincmn.cpp(372) 
> in CreateBase(): invalid id value

flixr, 2015-11-23:
thx @garciafa, can you please test the wxgtk3_fixes branch from #1438 ?

PR #1438 (2015-11-30): fixes for using wxPython 3.0 (python-wxgtk3.0)
Trying to fix #1435
closed 2015-11-30 by flixr