paparazzi/paparazzi #1443: ADC readings on Lisa MX are incorrect

podhrmic, 2015-11-25:
I just noticed after running ``tes_adc`` from ``test_progs`` on Lisa MX 2.1 that if I enable all 5 ADC channels (4xAUX, 1x V_BAT) and then have the input only on the V_BAT channel, I get readings in the unconnected AUX channels too, and they depend on V_BAT voltage, see below:
[image]

I noticed that I get the same behaviour with ChibiOS drivers (RT_Paparazzi). Is that normal? Is my board fried (I have only one to test on)? Is it a driver issue?

dewagter, 2015-11-25:
This is hardware characteristic of the processor. The ADC uses has a single
AD-channel with a multiplexer in front and a capacitor. So when it is
connected first to input 1 (VBat), then measured, then connected to input 2
(which was a floating input): then the voltage in capacitor will not change
a lot, but typically it will discharge a bit so a lower voltage is likely
to be measured. Then to 3, 4, 5... This is called cross talk and also
indicates that you need relatively low impedance analog inputs in order not
to have any cross-talk.

-Christophe

podhrmic, 2015-11-25:
Thanks for the explanation. Not an issue then.

closed 2015-11-25 by podhrmic