paparazzi/paparazzi #1444: MS5611 baro not working on Lisa MX

podhrmic, 2015-11-25:
With latest master and LIsa MX board, the baro doesn't work. The same configuration ``quadrotor_lisa_m_2_spektrum.xml`` works with Lisa M, but when I switch the board to ``lisa_mx_2.1`` in the airframe config file, I dont get any response from the baro.

Can someone verify if it is a problem with the code? I have only one Lisa MX to test on.

flixr, 2015-11-26:
Hm... It's working fine on my LisaMX 2.0... (tested with v5.7_devel-699-g20c4f93)
Don't have a LisaMX 2.1 to test...

podhrmic, 2015-11-26:
probably a HW problem then. 2.0 and 2.1 should be basically identical, right? Thanks for checking.

flixr, 2015-11-26:
@esden maybe you have a 2.1 around that you can quickly test?

podhrmic, 2015-12-03:
Confirmed, it was a hardware problem.

closed 2015-12-03 by podhrmic